MarlinFirmware/Marlin #18899: [BUG?] FSMC_GRAPHICAL_TFT on BOARD_MKS_ROBIN_NANO: Backlight pin redefined

Miraculix200, 2020-08-02:
Board: MKS Robin Nano (environment mks_robin_nano35)

When compiling the current bugfix version with #define TFT_LVGL_UI_FSMC, everything works. When I don't define TFT_LVGL_UI_FSMC but FSMC_GRAPHICAL_TFT, I get constant compiler warnings during compilation, like this:

```
Marlin/src/gcode/bedlevel/../../inc/../pins/stm32f1/pins_MKS_ROBIN_NANO.h:229:0: warning: "LCD_BACKLIGHT_PIN" redefined
 #define LCD_BACKLIGHT_PIN -1
 
Marlin/src/gcode/bedlevel/../../inc/../pins/stm32f1/pins_MKS_ROBIN_NANO.h:211:0: note: this is the location of the previous definition
 #define LCD_BACKLIGHT_PIN PD13
```

After restarting the printer it hangs at the "Booting..." screen. A serial connection can not be established.

I'm using a vanilla Configuration.h and pins_MKS_ROBIN_NANO.h, with just a few things changed in Configuration.h:

```
#define SERIAL_PORT 3
#define BAUDRATE 115200
#define MOTHERBOARD BOARD_MKS_ROBIN_NANO
#define FSMC_GRAPHICAL_TFT
```

The warnings will go away when I comment out the #if HAS_SPI_LCD section in pins_MKS_ROBIN_NANO.h, but it still hangs at the "Booting..." screen.

So why does it assume I have an SPI display? I didn't #define it anywhere. 

I'd call that a bug, because I'd expect to be able to choose between FSMC_GRAPHICAL_TFT and TFT_LVGL_UI_FSMC on the Robin Nano 1.2, not be forced to use TFT_LVGL_UI_FSMC. I mean it's a nice addition, but right now it's missing some menus I need.

I guess that this problem is also the reason why the screen won't work anymore with my non-vanilla bugfix-2.0.x (with pins file modified for UART etc.), starting with this commit: https://github.com/MarlinFirmware/Marlin/commit/38ce5966e19094fb499ce7d647961c0594c5f811

Before that commit I could simply comment out the #if HAS_SPI_LCD section to get rid of the warnings and it would work. The compiler warning spam started with my merge on June 25th, I guess it was caused by this commit (but I didn't try to pinpoint it):
https://github.com/MarlinFirmware/Marlin/commit/d20d4591321aa467c062f0050355d2aca84e03ad

rhapsodyv, 2020-08-02:
The commit you said just add support for nano 2 display (that is spi).
If it worked before, is because it use the “default” part in the display initialization. That’s the only modification was done that may impact your display.
I don’t have a nano 1 to test, but I can share a branch for you test it.

rhapsodyv, 2020-08-02:
Can you test this branch? https://github.com/rhapsodyv/Marlin/tree/fix-nano-v1-fsmc-display

If it work, I will send it in a PR

halilbil, 2020-08-02:
I have tried to compile this branch by uncommenting #define TFT_480x320 and #define TOUCH_SCREEN lines. But it failed with following error messages:

> In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfigPre.h:48:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/../HAL/STM32F1/inc/Conditionals_LCD.h:30:4: error: #error "Sorry! TFT displays are not available for HAL/STM32F1."
   #error "Sorry! TFT displays are not available for HAL/STM32F1."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfigPre.h:48:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:35:
Marlin\src\HAL\STM32F1\../../inc/../HAL/STM32F1/inc/Conditionals_LCD.h:30:4: error: #error "Sorry! TFT displays are not available for HAL/STM32F1."
   #error "Sorry! TFT displays are not available for HAL/STM32F1."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfigPre.h:48:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/../HAL/STM32F1/inc/Conditionals_LCD.h:30:4: error: #error "Sorry! TFT displays are not available for HAL/STM32F1."
   #error "Sorry! TFT displays are not available for HAL/STM32F1."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfigPre.h:48:0,
                 from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp:29:
Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/STM32F1/inc/Conditionals_LCD.h:30:4: error: #error "Sorry! TFT displays are not available for HAL/STM32F1."       
   #error "Sorry! TFT displays are not available for HAL/STM32F1."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfigPre.h:48:0,
                 from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp:21:
Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/STM32F1/inc/Conditionals_LCD.h:30:4: error: #error "Sorry! TFT displays are not available for HAL/STM32F1."       
   #error "Sorry! TFT displays are not available for HAL/STM32F1."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfigPre.h:48:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32F1\eeprom_bl24cxx.cpp:28:
Marlin\src\HAL\STM32F1\../../inc/../HAL/STM32F1/inc/Conditionals_LCD.h:30:4: error: #error "Sorry! TFT displays are not available for HAL/STM32F1."
   #error "Sorry! TFT displays are not available for HAL/STM32F1."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfigPre.h:48:0,
                 from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_hwspi.cpp:21:
Marlin\src\HAL\STM32F1\dogm\../../../inc/../HAL/STM32F1/inc/Conditionals_LCD.h:30:4: error: #error "Sorry! TFT displays are not available for HAL/STM32F1."       
   #error "Sorry! TFT displays are not available for HAL/STM32F1."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfigPre.h:48:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32F1\eeprom_flash.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/../HAL/STM32F1/inc/Conditionals_LCD.h:30:4: error: #error "Sorry! TFT displays are not available for HAL/STM32F1."
   #error "Sorry! TFT displays are not available for HAL/STM32F1."
    ^~~~~
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32F1\HAL.cpp.o] Error 1
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32F1\Servo.cpp.o] Error 1
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32F1\HAL_SPI.cpp.o] Error 1
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_fsmc.cpp.o] Error 1
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp.o] Error 1
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_hwspi.cpp.o] Error 1
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32F1\eeprom_bl24cxx.cpp.o] Error 1
*** [.pio\build\mks_robin_nano35\src\src\HAL\STM32F1\eeprom_flash.cpp.o] Error 1

rhapsodyv, 2020-08-02:
> #define TFT_480x320

This is not the same thing we are talking in this issue... `TFT_480x320` is the new UI, right now done only for STM32. 

Can you test with `FSMC_GRAPHICAL_TFT`?

halilbil, 2020-08-02:
Oh, I am sorry. I thought it is the same problem with that I am having. Because when I enabled `FSMC_GRAPHICAL_TFT`, I was also getting a lot of `#define LCD_BACKLIGHT_PIN -1 redefined` warnings during compilation. I got rid of these warnings by commenting `#define LCD_BACKLIGHT_PIN -1` line under `#if HAS_SPI_LCD` section of pins_MKS_ROBIN_NANO.h file. But this did not solve my problem. Marlin menu is still shown only on upper left corner of the lcd and it fills only two third of the screen.

Anyway, I tried to compile your branch by enabling `FSMC_GRAPHICAL_TFT` and disabling `TFT_480x320`. However compilation failed with following errors. I have attached my configuration file

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5012656/Configuration.zip)

> Marlin\src\feature\touch\xpt2046.cpp: In static member function 'static uint8_t XPT2046::read_buttons()':
Marlin\src\feature\touch\xpt2046.cpp:114:22: error: 'XPT2046_X_CALIBRATION' was not declared in this scope
       tsoffsets[0] = XPT2046_X_CALIBRATION;
                      ^~~~~~~~~~~~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:114:22: note: suggested alternative: 'XPT2046_CONTROL'
       tsoffsets[0] = XPT2046_X_CALIBRATION;
                      ^~~~~~~~~~~~~~~~~~~~~
                      XPT2046_CONTROL
Marlin\src\feature\touch\xpt2046.cpp:115:22: error: 'XPT2046_X_OFFSET' was not declared in this scope
       tsoffsets[1] = XPT2046_X_OFFSET;
                      ^~~~~~~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:115:22: note: suggested alternative: 'XPT2046_X'
       tsoffsets[1] = XPT2046_X_OFFSET;
                      ^~~~~~~~~~~~~~~~
                      XPT2046_X
Marlin\src\feature\touch\xpt2046.cpp:116:22: error: 'XPT2046_Y_CALIBRATION' was not declared in this scope
       tsoffsets[2] = XPT2046_Y_CALIBRATION;
                      ^~~~~~~~~~~~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:116:22: note: suggested alternative: 'XPT2046_CONTROL'
       tsoffsets[2] = XPT2046_Y_CALIBRATION;
                      ^~~~~~~~~~~~~~~~~~~~~
                      XPT2046_CONTROL
Marlin\src\feature\touch\xpt2046.cpp:117:22: error: 'XPT2046_Y_OFFSET' was not declared in this scope
       tsoffsets[3] = XPT2046_Y_OFFSET;
                      ^~~~~~~~~~~~~~~~
Marlin\src\feature\touch\xpt2046.cpp:117:22: note: suggested alternative: 'XPT2046_Y'
       tsoffsets[3] = XPT2046_Y_OFFSET;
                      ^~~~~~~~~~~~~~~~
                      XPT2046_Y
*** [.pio\build\mks_robin_nano35\src\src\feature\touch\xpt2046.cpp.o] Error 1

rhapsodyv, 2020-08-02:
> > XPT2046_Y
> > *** [.pio\build\mks_robin_nano35\src\src\feature\touch\xpt2046.cpp.o] Error 1

The compilation is just telling that you need those configurations
I dont the values for nano v1.2, but this are the values for nano 2, try it:
```
  #define XPT2046_X_CALIBRATION   -5481
  #define XPT2046_Y_CALIBRATION   4000
  #define XPT2046_X_OFFSET        343
  #define XPT2046_Y_OFFSET        0
  #define XPT2046_Z1_THRESHOLD    0
```

thisiskeithb, 2020-08-02:
This is what's used in the [Tevo Nereus config](https://github.com/MarlinFirmware/Configurations/blob/bb29dafbd0046a88389541203c5b98a3800a4fe0/config/examples/Tevo/Nereus/Configuration.h#L2195-L2206) which uses a Robin Nano 1.2:

```cpp
#define TOUCH_SCREEN
#if ENABLED(TOUCH_SCREEN)
  #define BUTTON_DELAY_EDIT  50 // (ms) Button repeat delay for edit screens
  #define BUTTON_DELAY_MENU 250 // (ms) Button repeat delay for menus

  #define TOUCH_SCREEN_CALIBRATION

  #define XPT2046_X_CALIBRATION  12149
  #define XPT2046_Y_CALIBRATION  -8746
  #define XPT2046_X_OFFSET         -35
  #define XPT2046_Y_OFFSET         256
#endif
```

rhapsodyv, 2020-08-02:
> This is what's used in the [Tevo Nereus [...]

If those values work in nano v2 fsmc, I will put it as default in the board config. I think it's better for the users.
What do you think?

thisiskeithb, 2020-08-02:
I don't have a V2 yet to test with, so I don't know.

Miraculix200, 2020-08-02:
> Can you test this branch? https://github.com/rhapsodyv/Marlin/tree/fix-nano-v1-fsmc-display
> 
> If it work, I will send it in a PR

Well, it does compile now without the "backlight pin redefined" spam. But it doesn't produce a functional display. I only see the touch buttons. The upper half of the display is black, the lower part is white.

 I'll have a look at the pins file to see if I can change something.

thisiskeithb, 2020-08-02:
> But it doesn't produce a functional display. I only see the touch buttons. The upper half of the display is black, the lower part is white.

I'm seeing (hah!) the same thing: https://github.com/MarlinFirmware/Marlin/pull/18901#issuecomment-667716368

Miraculix200, 2020-08-02:
Still the same problem with the latest commit to https://github.com/rhapsodyv/Marlin/tree/fix-nano-v1-fsmc-display

Trying again with LCD_RESET_PIN commented out

Miraculix200, 2020-08-03:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/18901

PR #18901 (2020-08-04): Fix MKS Nano V1.2 FSMC Display
### Description

- Fix compiling warnings and the FSMC display for MKS Nano v1.2
- Add 3x upscale support for NON-DMA TFT
- Fix the 0x8066 init code
- Removed non supported displays that were add to nano board by mistake
- Re-enabled DMA for FSMC_GRAPHICAL_TFT for nano v1.2

### Benefits

Fix #18899 

### Related Issues
closed 2020-08-03 by Miraculix200