MarlinFirmware/Marlin #18900: [BUG] Inconsistencies in autobuild/dependencies script

Marlor, 2020-08-02:
Found 2 potential bugs in the dependencies script/autobuild

This is the output of a cleen unchanged marlin bugfix-2.0.x configuration:

```
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
[...]
```

Found 5 and only using 4?

Only changing `MOTHERBOARD` to `BOARD_BTT_SKR_V1_3` gives this output:

```
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <Adafruit NeoPixel> 1.5.0
Building in release mode
[...]
```

Found 3 and using 3, but is NeoPixel always needet on an SKR V1.3?

ellensp, 2020-08-02:
If don't get libraries it needs it will fail to compile.
If you add in extra they are just ignored. 

In the case of the LPC1768/9 it needs a special version of neopixel library. 
So it is always added in case it is needed. 
It is in the platformio.ini 
[common_LPC] 
lib_deps            Adafruit NeoPixel=https://github.com/p3p/Adafruit_NeoPixel/archive/1.5.0.zip

If this was left to the auto detection script it would only include neopixel library when it is needed, but it would use the standard neopixel library that doesn't work on lpc1768/9

These are questions. not bugs..

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-08-02 by ellensp

Marlor, 2020-08-02:
> 
> These are questions. not bugs..

The feature: https://github.com/MarlinFirmware/Marlin/pull/18699 "Optimize build. Only used libraries downloaded & compiled."
The reality: downloades & compiles unused libraries on lpc1768/9 
Please tell me you can see how this can be interpreted as a bug

Its a new feature so i rather reported than ignore it.
These reports are not welcome here? Maybe you should add this info in the docs

ellensp, 2020-08-02:
you did read my reply

the LPC1768 overrides this new feature for neopixels, as it has to!

Marlor, 2020-08-02:
I know, that wasn't what my reply was about

thinkyhead, 2020-08-04:
> "Found 5 compatible libraries"

A message from PlatformIO, perhaps …before… the processing script is run?

Anyway, if the code is compiling then it's not a big worry. We'll keep tweaking the dependency script, and we should have some new changes that will allow us to handle dependencies by feature inside of individual environments, the same way we now do at the global scope.

More thrilling discussions about PlatformIO advanced scripting are ongoing in the Discord, so drop by anytime.