MarlinFirmware/Marlin #18905: Marlin 2.0.6 [BUG] G29 Starts from the end of X axis

insane78, 2020-08-02:
### Bug Description

I have same issue with Marlin 2.0.6 and the last bugfix.
I'm using a sidewinder x1 with SKR 1.4 Turbo and 3.5 touch BTT 
For leveling i'm using Bilinear Auto with 4 Grid Points and original BLTouch 3.1 used as servo (not as endstop)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5012491/Marlin.zip)

### Steps to Reproduce

1- G28
2- G29

**Expected behavior:** [What you expect to happen]
I'm expectig that the probe starts at X:0 and Y=0 ( the first point of the mesh grid) 

**Actual behavior:** [What actually happens]
The probe starts at X:Max and Y:0
This cause inverted Z compensation on X Axis during print, i have areas wich the nozzle have to be lowered that are not touching the bed, and other areas that are closer to nozzle, where the nozzle are squished more. 
So i have the nozzle far away from lower areas and nearest to higher areas.

bill-orange, 2020-08-02:
On your SKR board, try plugging the stepper  in Z0 into Z1 and the stepper plugged into Z1 into Z0.

insane78, 2020-08-03:
> On your SKR board, try plugging the stepper in Z0 into Z1 and the stepper plugged into Z1 into Z0.

Z1 and Z0 moving at same time, there's no difference swapping them... My problem is that the grid is taken from right to left, end not left to right, so mi first matrix point is the X MAX position and not the X MIN position.....

sjasonsmith, 2020-08-03:
How many Y probe points are you usin? Starting on the right side is expected if it is an even number.

insane78, 2020-08-03:
> How many Y probe points are you usin? Starting on the right side is expected if it is an even number.

Same behaviour with 4x4 grid and 6x6 grid

sjasonsmith, 2020-08-03:
> Same behaviour with 4x4 grid and 6x6 grid

Those are both even numbers. Try with 3 or 5 and you will see it starts on the left side. This is expected.

closed 2020-08-05 by insane78