MarlinFirmware/Marlin #18908: [BUG] M871 command miss first index value

BastR, 2020-08-03:
### Bug Description

I try to enter manually (with the 871 command) the compensation value for the pinda probe but I can't specify the first value of the table.  

Index 1 (I1) is always the second value of the table, I tried I0 just in case, but it's not possible.

[image]

@tompe-proj

boelle, 2020-08-03:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

BastR, 2020-08-03:
My config is here: https://github.com/BastR/Marlin/tree/V-Core-Pro/Marlin

TheNeskik, 2020-08-03:
If I'm not wrong, the temp probe compensation need to have your Z-offset well calibrate before doing G76.
When you do your Z-offset calibration, you need to do it with your bed set to the first temperature of your bed compensation and your probe set to the first value of the table. (Possibly, we can admit that below 50°C for the bed and 30°C for the probe their is no shift). That's why your index 0 (I0) can't be edit because you have already calibrate your printer for this temp.

that's what i always thought

BastR, 2020-08-03:
That makes sense, but I'd like confirmation. 

My Z-offset is well calibrated, however I can't get the automatic calibration (G76) to work, I tried it 3 times and each time the process goes on hold and I have to reset the machine. It's quite hot at home at the moment, I don't know if it's because of that, that's why I wanted to enter the compensation values myself using the table of an original pinda V2 provided here: https://marlinfw.org/docs/features/probe_temp_compensation.html

rjfsilva, 2020-08-17:
I have not tried to enter a compesation value manually yet, but I can confirm the same behaviour when I try the automatic calibration (G76).