MarlinFirmware/Marlin #18910: Error compiling in 2.0.6 due to G35.cpp and ASSISTED_TRAMMING.

jariivanoff, 2020-08-03:
### Bug Description

Get this error when compiling 2.0.6 ( also with Bugfix version ), for SKR 1.4 Turbo and SKR 1.3 ( haven't tested for other boards )
Compiling .pio\build\LPC1769\src\src\gcode\calibrate\M425.cpp.o

Marlin\src\gcode\bedlevel\G35.cpp: In static member function 'static void GcodeSuite::G35()':

Marlin\src\gcode\bedlevel\G35.cpp:126:41: error: 'class Probe' has no member named 'probe_at_point'

126 | const float z_probed_height = probe.probe_at_point(screws_tilt_adjust_pos[i], PROBE_PT_RAISE, 0, true);

| ^~~~~~~~~~~~~~

Marlin\src\gcode\bedlevel\G35.cpp:126:83: error: 'PROBE_PT_RAISE' was not declared in this scope

126 | const float z_probed_height = probe.probe_at_point(screws_tilt_adjust_pos[i], PROBE_PT_RAISE, 0, true);

| ^~~~~~~~~~~~~~

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Enable ASSISTED_TRAMMING

**Expected behavior:** 
Compiling would go through and you get the Assisted Tramming functionality

**Actual behavior:** 
Compilation fails:
Compiling .pio\build\LPC1769\src\src\gcode\calibrate\M425.cpp.o

Marlin\src\gcode\bedlevel\G35.cpp: In static member function 'static void GcodeSuite::G35()':

Marlin\src\gcode\bedlevel\G35.cpp:126:41: error: 'class Probe' has no member named 'probe_at_point'

126 | const float z_probed_height = probe.probe_at_point(screws_tilt_adjust_pos[i], PROBE_PT_RAISE, 0, true);

| ^~~~~~~~~~~~~~

Marlin\src\gcode\bedlevel\G35.cpp:126:83: error: 'PROBE_PT_RAISE' was not declared in this scope

126 | const float z_probed_height = probe.probe_at_point(screws_tilt_adjust_pos[i], PROBE_PT_RAISE, 0, true);

| ^~~~~~~~~~~~~~

#### Additional Information
I have tried with different settings for the tramming etc. for 2 cards with the same result. With both stable 2.0.6 and Bugfix downloaded on August 2nd. Only way to get a successful compilation is to disable ASSISTED_TRAMMING again.

qwewer0, 2020-08-03:
> **Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

boelle, 2020-08-03:
@jariivanoff 

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

jariivanoff, 2020-08-03:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5016346/Marlin.zip)
Oh.. sorry... forgot those... Here they are... only difference between working and not working compilation is  #define ASSISTED_TRAMMING being enabled or disabled. Have tried with "stock" settings for "corner points" which are for a 200x200mm bed and with adjusted values for my 310x320 bed.

qwewer0, 2020-08-03:
Do you have bed leveling sensor? `ASSISTED_TRAMMING` can only work with an ABL.

jariivanoff, 2020-08-03:
Ahh.... no I have not... I thought it was like this: You get the printhead sent to the 4 corners and lowered so you can adjust the bed manually. I have a similar function in my BTT TFT35 screens, where it placese the head in each of the corners of choice. I have similar in FW for my Tevo Tornados under custom commands ( which are on stock FW ), so I thought it was that function. Send head to front right, front left, rear right and rear left of the bed.

thisiskeithb, 2020-08-03:
> I thought it was like this: You get the printhead sent to the 4 corners and lowered so you can adjust the bed manually.

Enable `LEVEL_BED_CORNERS` for that functionality.

jariivanoff, 2020-08-03:
Ahh... thanks... my bad..

thisiskeithb, 2020-08-03:
I added a sanity check for this feature in https://github.com/MarlinFirmware/Marlin/pull/18912.

closed 2020-08-03 by thisiskeithb