MarlinFirmware/Marlin #18911: [BUG] Marlin does not execute the second filament runout script when using a mixing extruder

Hoodie60, 2020-08-03:
### Bug Description

Hi everyone,

I´m having trouble getting a second filament runout sensor for my system to run. I´m using a mixing extruder consisting of two extruders. On M119 both runout sensors show up correctly. They get triggered as planned, but only the script of the first sensor gets executed. Marlin runs on a RAMPS 1.4 Board.
Am I doing something wrong or is this a bug in Marlin?

Thanks in advance!

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5016573/configs.zip)

Cura start G-code:

M163 S0 P0.5
M163 S1 P0.5
M164 S3
T3

### Steps to Reproduce

1. Configure Marlin version 978e0e4 with the attached files

2. Hook up two filament runout sensors & trigger them separately

**Expected behavior:** Both runout scripts should be executed when their associated sensor gets triggered.

**Actual behavior:** Only the first script gets executed (I've even had an older bugfix version where both sensors triggered the first runout script)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-08-04:
so you are running  [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

Hoodie60, 2020-08-04:
No, I've also tried the bugfix-2.0.x branch, but at the moment I`m running the 2.0.x branch.

Hoodie60, 2020-09-04:
Hi, were you able to reproduce this issue?
Thanks in advance!

closed 2020-09-23 by github-actions[bot]