MarlinFirmware/Marlin #18913: 2.0.6 doesn't compile with Arduino 1.9.0 IDE

T69, 2020-08-03:
When compiling 2.0.6 with Arduino 1.9.0 Beta IDE I get the following errors (Compiling 2.0.4 is fine):

In file included from sketch\src\inc/MarlinConfig.h:30:0,

                 from sketch\src\MarlinCore.h:24,

                 from sketch\src\MarlinCore.cpp:31:

sketch\src\inc/../HAL/HAL.h:41:46: error: redeclaration 'NUL_STR' differs in 'constexpr'

   #define PGMSTR(NAM,STR) constexpr char NAM[] = STR

                                              ^

sketch\src\MarlinCore.cpp:211:1: note: in expansion of macro 'PGMSTR'

 PGMSTR(NUL_STR, "");

 ^

In file included from sketch\src\MarlinCore.cpp:31:0:

sketch\src\MarlinCore.h:122:19: error: from previous declaration 'NUL_STR'

 extern const char NUL_STR[], M112_KILL_STR[], G28_STR[], M21_STR[], M23_STR[], M24_STR[],

                   ^

In file included from sketch\src\inc/MarlinConfig.h:30:0,

                 from sketch\src\MarlinCore.h:24,

                 from sketch\src\MarlinCore.cpp:31:

sketch\src\inc/../HAL/HAL.h:41:46: error: redeclaration 'M112_KILL_STR' differs in 'constexpr'

   #define PGMSTR(NAM,STR) constexpr char NAM[] = STR

                                              ^

sketch\src\MarlinCore.cpp:212:1: note: in expansion of macro 'PGMSTR'

 PGMSTR(M112_KILL_STR, "M112 Shutdown");

 ^

In file included from sketch\src\MarlinCore.cpp:31:0:

sketch\src\MarlinCore.h:122:30: error: from previous declaration 'M112_KILL_STR'

 extern const char NUL_STR[], M112_KILL_STR[], G28_STR[], M21_STR[], M23_STR[], M24_STR[],

                              ^

In file included from sketch\src\inc/MarlinConfig.h:30:0,

                 from sketch\src\MarlinCore.h:24,

                 from sketch\src\MarlinCore.cpp:31:

sketch\src\inc/../HAL/HAL.h:41:46: error: redeclaration 'G28_STR' differs in 'constexpr'

   #define PGMSTR(NAM,STR) constexpr char NAM[] = STR

                                              ^

sketch\src\MarlinCore.cpp:213:1: note: in expansion of macro 'PGMSTR'

 PGMSTR(G28_STR, "G28");

[... 506 lines cut ...]
                 from sketch\src\MarlinCore.h:24,

                 from sketch\src\MarlinCore.cpp:31:

sketch\src\inc/../HAL/HAL.h:41:46: error: redeclaration 'SP_E_LBL' differs in 'constexpr'

   #define PGMSTR(NAM,STR) constexpr char NAM[] = STR

                                              ^

sketch\src\MarlinCore.cpp:222:76: note: in expansion of macro 'PGMSTR'

 PGMSTR(SP_X_LBL, " X:"); PGMSTR(SP_Y_LBL, " Y:"); PGMSTR(SP_Z_LBL, " Z:"); PGMSTR(SP_E_LBL, " E:");

                                                                            ^

In file included from sketch\src\MarlinCore.cpp:31:0:

sketch\src\MarlinCore.h:125:91: error: from previous declaration 'SP_E_LBL'

                   X_LBL[], Y_LBL[], Z_LBL[], E_LBL[], SP_X_LBL[], SP_Y_LBL[], SP_Z_LBL[], SP_E_LBL[];

Is Platformio now the preferred IDE environment or is there any way to fix this?

Thanks & Regards

Grogyan, 2020-08-03:
Marlin is intended to be compiled with Platformio with VSCode

boelle, 2020-08-03:
@T69 

> Is Platformio now the preferred IDE environment

Yes

ellensp, 2020-08-03:
Arduino 1.9.0 beta build (not hourly), marlin 2.0.6. Default RAMPS configs. on Linux...  compiles without issues.

ellensp, 2020-08-04:
Just tried the same on windows. Also compiled without issues.

Cannot reproduce.

T69, 2020-08-04:
Thanks ellensp! 
Installed Arduino 1.9.0 beta build (not hourly) & additional libraries ...

Compiling for MOTHERBOARD BOARD_RAMPS_14_EFB works ok ...
Compiling for MOTHERBOARD BOARD_RAMPS4DUE_EFB is apparently the problem and gives me the errors above ...

boelle, 2020-08-04:
@T69 

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

T69, 2020-08-04:
I'm sorry, configuration is now attached ... thanks for looking into this ...

[conf.zip](https://github.com/MarlinFirmware/Marlin/files/5021934/conf.zip)

ellensp, 2020-08-05:
I can confirm there is an issue. but i'm not sure if its a arduino issue or a marlin issue.  Since it compiles fine on other platforms I a inclined to believe its an arduino ide issue. 
The final nail in the coffin for arduino ide perhaps?

T69, 2020-08-05:
Just compiled 2.0.5 for BOARD_RAMPS4DUE_EFB without issues ...

T69, 2020-08-05:
Compiled 2.0.6 with same configuration using PlatformIO was working, but after uploading the REPRAP_DISCOUNT_SMART_CONTROLLER is dead/not showing anything, but OctoPrint can connect to 2.0.6 via USB ...

AFprinter, 2020-08-08:
Is this for an archim board?

T69, 2020-08-08:
No, it's for an arduino due with a modded RAMPS 1.5 ...

closed 2020-08-24 by ellensp

thinkyhead, 2020-08-26:
> is it possible to update the firmware from SD card in case of 32-bit boards

For some 32-bit boards there is no USB update option at all, only SD card updates or direct linking to a pin header with a debugger device.