MarlinFirmware/Marlin #18914: [FR] Make confirmation dialog for starting and stopping a print configurable

nilshott, 2020-08-03:
### Description

Since version 2.0 there is a confirmation dialog when starting or stopping a print. I tried to get used to it, but my brain refuses to. So can we please add a way to configure this behavior?

### Example for the _Stop Print_ menu item:

The line `#define CONFIRM_STOP_PRINT true` could produce the current code
`MENU_ITEM(submenu, MSG_STOP_PRINT, menu_abort_confirm);`
whereas the line `#define CONFIRM_STOP_PRINT true` could produce the previous code
`MENU_ITEM(submenu, MSG_STOP_PRINT, []{ ui.abort_print() ; ui.goto_previous_screen(); });`

Likewise for _Start Print_ and other potentially dangerous commands.

qwewer0, 2020-08-03:
`SD_MENU_CONFIRM_START`             // Confirm the selected SD file before printing

For aborting a print job without confirming it, I think that is a really bad idea. Just think about hitting cancel by accident in the middle of a 24 h print.

thinkyhead, 2020-08-04:
Marlin is open source exactly so that you can make that change, but I don't think it should be exposed as a general option. An emergency stop attached to a pin of its own would be a proper alternative.

nilshott, 2020-08-08:
Thanks guys,
now I have everything I need to set up Marlin is I want without it needing to be a feature request.

closed 2020-08-08 by nilshott