MarlinFirmware/Marlin #18916: [BUG] PID Autotune displays garbage on status area

nicwilson58, 2020-08-04:
When PID Autotune is enabled and used,  or a M303 command is issued,  the command works as expected but displays garbage on  the status area and remains there during the process.  Once the process is complete,  the status returns to normal and can be written with M117

ellensp, 2020-08-04:
Please attach your configuration files.  We need to know what type of display, controller, marlin version and possibly other setting. These can be obtained from the config files.
Perhaps also attach an image of the garbage on the display.

boelle, 2020-08-04:
@nicwilson58 

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

nicwilson58, 2020-08-07:
Config files, and display photos attached as requested.  here is a step by step procedure of the issue,  doing these one at a time, shows the issue lies within the M303 command itself when issued together in a custom user menu,  it happens so quick the first thing you see is the garbage.

1.  issue  M117 Start PID
2. Status shows Start PID
3. issue M106 S128
4. display shows fan 50%
5. issue M303 C8 S200 E0 U
6. PID process begins normally,  but status text is overwritten with garbage.  (pic attached)
7 on completion issue M117 PID Completed.
8. display shows PID Completed.

I have also added these commands to a user menu.  you never see the first correct text as its very quickly overwritten with the garbage of the M303 command, on completion though the final M117 text is shown normally, proving once more the garbage is being generated in the M303 commands.

[image]

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5038849/Config.zip)

sjasonsmith, 2020-08-17:
I'm using an SKR Pro and not a stock control board, but I'm not seeing any issues like this.

It is clear from your configuration files that you are not using stock Marlin. We can't support Marlin forks from here. Please try to reproduce the issue using Marlin directly. There is most likely an example configuration available for your printer.

Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

nicwilson58, 2020-08-17:
Wrong,  the Marlin IS stock,  I wrote the few extra defines to take care of options,  however removing them shows the exact same issue,  the configuration.h CANNOT CAUSE such an issue like this.

boelle, 2020-09-11:
So you did retest with [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

nicwilson58, 2020-09-11:
no but I willSent from my Samsung Galaxy smartphone.

nicwilson58, 2020-09-12:
I just tested with the bug fix branch and the LCD  garbage is still there on the start of PID tune.  For those who think its my branch I also compiled a vanilla standard build with only printed dimensions and board changed as little as I could  and it still shows the garbage when Auto PID tune starts,  and correct text on PID autotune end.

 

nicw

sjasonsmith, 2020-09-12:
Is the garbage always the same or is it random?

nicwilson58, 2020-09-13:
Its Random, at the start of the PID code the source is doing a screen print, I expect it pointing to unset variable or such.

nicwilson58, 2020-10-26:
Under the latest V2..0.7.2 of Marlin,  its no longer displaying garbage,  but the status text line is cleared,  it should remain untouched.

nicwilson58, 2020-10-26:
Likewise,  when running PID from the Marlin menu,  the first thing it does is send "PID Autotune start" to the status,  this is never seen though as its immediately cleared by the PID autotune code.

sjasonsmith, 2020-11-10:
@nicwilson58 can you verify whether you still have any issues?
When I run a PID tune, I see `PID Autotune` in the display for the duration of the operation.

Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

boelle, 2020-11-10:
i see the same as @sjasonsmith

nicwilson58, 2020-11-10:
Hi,

I will test ASAP

Nic

nicwilson58, 2020-11-11:
hi,
 
I can build the main branch fine, but I cannot build the current bug fix due to an error:
 
src\lcd\menu\menu.cpp:295:30: error: 'manual_feedrate_mm_s' was not declared in this scope
 
This seems to be with "MANUAL_FEEDRATE" which is not being used.   As this is not an issue with the main branch I am wondering if there is a bug in the bugfix version.
 
 
Nic

nicwilson58, 2020-11-14:
I managed to fix the compile error,  and such changes really annoy me as a programmer since 1980,  our changes to the configuration files are hard enough to roll in new versions without the unnecessary changing of #define names,  its a useless practise and does not improve the code when they are quite readable before and well known.  Now my rant is over,  the test result was the same,  the Configuration-Advanced Settings-Temperature-PID works fine,  but the lower left status area is blank,  so still being overwritten.

sjasonsmith, 2020-11-15:
@nicwilson58 can you post a picture of what you see on your screen, and describe what you expect to see?

Please also post the configuration files you are using with `bugfix-2.0.x`. This might be a problem which impacts one platform but not others.

rhapsodyv, 2020-12-02:
Can you share your updated configs? 
For me it's working, after pid autotune finishes, it shows the text stored in: `MACHINE_NAME`, that can be "3D Printer", or the contents of `CUSTOM_MACHINE_NAME`.

If you edit `CUSTOM_MACHINE_NAME`, the problem continues?

sjasonsmith, 2020-12-02:
If we don't hear back within a day or two from @nicwilson58 I think we just need to close this issue. It isn't clear from the last post what the remaining issue is. I'm not sure what "the lower left status area is blank" refers tom versus what is expected to be seen.

nicwilson58, 2020-12-02:
hi 
 
please do not assume everyone here is home all the time.  I travel 70% of the  year for work and Dec is out busiest time,  and I am not home to test right now.  this is still an issue.   Mine also DOES show text when PID finishes as my original post says,  but when it starts it shows nothing and its not even clear the process has begun until you see hotend temp start to rise.

sjasonsmith, 2020-12-03:
> please do not assume everyone here is home all the time.

Well, we had been waiting 18 days for a reply...

closed 2021-01-03 by rhapsodyv