MarlinFirmware/Marlin #18917: [BUG] M166 (Gradient Mix) uses Z tower stepper position on delta

Kadah, 2020-08-04:
### Bug Description

M166 is seemingly using the Z tower stepper position to calculate the gradient factor.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5021777/Marlin.zip)

### Steps to Reproduce

1. G28 ;print head to top of printer makes it easier to spot disparity
2. M166 ;Note reported Current Z value
3. G0 Y-40
4. M166 ;Note and compare new reported Current Z value to previous

**Expected behavior:** [What you expect to happen]

In both M166 calls, the reported Current Z should be the current tool Z position and should not change when only moving on Y axis.

**Actual behavior:** [What actually happens]

The reported Current Z is the position of the Z tower slider. When moving negative on Y, only the Z tower axis moves downward and the reported value (which is what is used for the gradient updates) changes.

M166 command on a detla would not have the expected result, and would mostly only use the final vtool mix.

Homed:
Gradient Mix ON ; Start Z0.10 T0 (100%|0%) ; End Z5.50 T1 (0%|100%) ; **Current Z537.29** (0%|100%)
Home Y-40:
Gradient Mix ON ; Start Z0.10 T0 (100%|0%) ; End Z5.50 T1 (0%|100%) ; **Current Z508.67** (0%|100%)
0,0,0:
Gradient Mix ON ; Start Z0.10 T0 (100%|0%) ; End Z5.50 T1 (0%|100%) ; **Current Z209.46** (0%|100%)
0,-40,0:
Gradient Mix ON ; Start Z0.10 T0 (100%|0%) ; End Z5.50 T1 (0%|100%) ; **Current Z171.47** (0%|100%)

#### Additional Information

Running Marlin-bugfix-2.0.x from 2020-7-18.

Hotend currently is not wired or pushing plastic yet, but in air printing, I can see that an 'M166 A0.1 Z5.5 I0 J1 S1' is only doing the final vtool mix since the lowest the Z tower can go on the printer is around 170mm when at the every front of the print bed. I suspect if it were to print for real with this, the effect would be a rather interesting spheroid gradient towards the rear.

I was going to test fixing this, but I got as far as Planner::get_axis_position_mm() does not translate for deltas and felt that it was best that leave that PR for somebody more familiar with Marlin's code.

Kadah, 2020-08-06:
I don't know if the below changes are "correct" way (feels like hack), the console output is what I would expect now. Will test for real once the printer able to melt (waiting on parts to finish wiring the hotend).

Marlin\src\feature\mixing.cpp
```
  void Mixer::update_gradient_for_planner_z() {
    get_cartesian_from_steppers();
    update_gradient_for_z(cartes.z);
  }
```

Marlin\src\gcode\feature\mixing\M166.cpp
```
    mixer.update_mix_from_gradient();
    SERIAL_ECHOPAIR(" ; Current Z", cartes.z);
    echo_mix();
  }
```

sjasonsmith, 2020-08-09:
This isn't a feature I've ever used, but I suspect you are at least close to a proper solution. I suggest you post a pull-request with the change. Thinkyhead will probably look at it and immediately know whether this is correct or not.

Kadah, 2020-08-10:
Was going to make it a PR when I get the printer up to actually printing.

Ran in to some setback and had to replace the controller with a completely different one earlier than expected (The Fysetc F6 1.3 has too many hardware bugs, then some internal pullups died. Replacing with BTT SKR Pro because Amazon Prime).

I want to see what the effect of the existing behavior is on a print, I suspect it could be interesting, and that gives me ideas,

Kadah, 2020-09-06:
Finally got those demos printed (printer had a number of issues to sort out). Right is stock 2.0.6.1, left with patch. Gcode is the same with the exception that the left one has the M166 linears +200mm to compensate for the tangent offset.

The gocde has 3 M166 sections to make gradients between 4 colors, so the gradient gets cut off on the stock 2.0.6.1 print when it gets to the next section. There's an underextrusion at the start of those sections (more visible on the left face going in to the violet), that might be due to going below the A<linear>, have not checked.

[image]

sjasonsmith, 2020-09-06:
@Kadah thanks for posting pictures! Are you still planning on posting a pull request yourself to fix it?

Kadah, 2020-09-06:
@sjasonsmith Yes, will likely have time to get that done later today. Will also look at if I can wrap my head around the math to make a z only forward kinematic and save some cycles.

closed 2020-10-14 by github-actions[bot]