MarlinFirmware/Marlin #18921: [FR] FIL_RUNOUT2_PIN for CHITU_V6

Masanetz, 2020-08-05:
Please add

`#define FIL_RUNOUT2_PIN      PF13`

to src/pins/stm32f1/pins_CHITU3D_V6.h
for support of second filament runout sensor of "Tronxy X5SA 2E"

Tested here with bugfix-2.0.x (2020-08-04)

thisiskeithb, 2020-08-05:
I put in https://github.com/MarlinFirmware/Marlin/pull/18922 to add the second filament runout pin.

Does the pin/connector have a label on the motherboard?

Masanetz, 2020-08-05:
Should be FILAMENT2 as on [picture of board at tronxyonline.com](https://www.tronxyonline.com/Tronxy-Silent-Mainboard-with-Wire-Cable-for-X5SA-Series-and-XY-2-Pro-p219681.html), but I have to check in my printers control box after longer running print...

Masanetz, 2020-08-06:
Just checked the board, it's FILAMENT2...

thisiskeithb, 2020-08-06:
Closing since the PR was merged and the second filament sensor should now be usable.

closed 2020-08-06 by thisiskeithb