MarlinFirmware/Marlin #18923: [FR] Feed Rate Unit

qwewer0, 2020-08-05:
### Description

Is there a way to make `MANUAL_FEEDRATE` in Configuration_adv.h to be in mm/s instead of mm/min, like it is for `DEFAULT_MAX_FEEDRATE` and `MAX_FEEDRATE_EDIT_VALUES` in Configuration.h?
Or to clarify the used unit.

e.g.
`MANUAL_FEEDRATE { 50*60, 50*60, 4*60, 60 }` -> `MANUAL_FEEDRATE { 50, 50, 4, 1 }`
or
`Feedrates for manual moves along X, Y, Z, E from panel` -> `Feedrates (mm/min) for manual moves along X, Y, Z, E from panel`

sjasonsmith, 2020-08-09:
I agree that this is confusing.

thinkyhead, 2020-08-11:
You cannot just change the units of an existing configuration option. It would need to be renamed (e.g., `MANUAL_FEEDRATE_MM_S` if new units are going to be used. And then I suppose the longstanding and venerable `DEFAULT_MAX_FEEDRATE` would need to be renamed to `DEFAULT_MAX_FEEDRATE_MM_S`…. And then it's no use. All the rest of the feedrate settings will need to be likewise changed.

But then… When you consider that the feedrate `F` parameter in all of our G-code is always in current-units-per-_minute_, then maybe all settings should really be converted to mm/m for overall consistency. And, there are several other G-codes which I am sure take a feedrate parameter in current-units-per-_second_, so I guess it would be pretty disruptive to change all of those.

In general, it seems that each feature was defined in an _ad hoc_ way, as befit its author at the time it was implemented. In some cases users may find it easier to think in terms of mm/m because they set their movement speed, accelerations, and limits according to those numbers. The use of `n*60` values in a feedrate setting is more about providing a coarse adjustment for those values, but perhaps they should just be set as whole numbers so they correlate with the common experience of `F<feedrate>` parameters and other settings like it which are expressed in units-per-minute…?

qwewer0, 2020-08-11:
Then the easiest option would be to just add the unit in the comment to notify the user.
`#define MANUAL_FEEDRATE { 50*60, 50*60, 4*60, 60 } // Feedrates (mm/min) for manual moves along X, Y, Z, E from panel`

thinkyhead, 2020-08-11:
Thanks! All items that aren't clear should at least have a comment about what their units are.

closed 2020-08-11 by thinkyhead

AnHardt, 2020-08-11:
Yes. But here the notation with "x*60" already was a strong indicator.

qwewer0, 2020-08-11:
Better safe than sorry.