MarlinFirmware/Marlin #18924: avr-gcc: error: CreateProcess: No such file or directory[BUG] (short description)

flivermo, 2020-08-05:
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5028328/Configs.zip)

### Bug Description

When trying to upload the sketch i get the avr-gcc: error: CreateProcess: No such file or directory error.  I Tried several of the fixes I found on here and the only progress i can make is by removing some of the HAL files.  When i remove them everything works expect I am getting an error uploading sketch.  I am a first time user so I apologize if I am just missing something simple.

### My Configurations

Configurations attached

### Steps to Reproduce

1. Attempt to upload the sketch
2. Get error compiling
3. Remove some HAL files (I am new to this and do not know which ones I can remove)
4. Attempt to upload and get error uploading

**Expected behavior:** [What you expect to happen]

I would expect to get something saying it was uploaded successfully and when i restart the board I would prefer to see the boot screen.

**Actual behavior:** [What actually happens]

I get an error and then nothing.  Still have a blue screen.  I have even tried swapping the LCD wires as suggested in other places but no luck.

boelle, 2020-08-05:
have you tried vscode+platform.io ? arduino ide is not used much anymore and its filled with bugs that cause compiles to fail

flivermo, 2020-08-05:
I have not but will try to muddle through it tonight.  Are they any instructions available on how to upload to a Rambo 1.4 board or is it pretty self explanatory?

Thanks

ellensp, 2020-08-05:
the main gotcha with the rambo is "You must connect an external power source to the Heater, Fans, and Logic input for the RAMBo to operate. "   If you just use a usb cable only the usb chip is get power, not the processor.

closed 2020-08-06 by ellensp