MarlinFirmware/Marlin #18926: [FR] G35 TRAMMING_POINT_XY Sanity Check

qwewer0, 2020-08-05:
### Description

It would be useful to have a sanity check for `G35` `TRAMMING_POINT_XY`, to check if the probe can reach all the points or not.
It could be calculated from `NOZZLE_TO_PROBE_OFFSET`, `X_MIN_POS`, `Y_MIN_POS`, `X_MAX_POS`, `Y_MAX_POS` and `TRAMMING_POINT_XY`.

sjasonsmith, 2020-08-09:
You can't make a perfect sanity check for this, since the probe offsets can be modified at runtime.

For G34 checks were added that at least verify the _default_ probe offset and probing margin are compatible with the selected points.

Here is a link to that source which can serve as an example:
https://github.com/MarlinFirmware/Marlin/blob/8318d90e859219af9aa750ba2b91c7568f4dac55/Marlin/src/feature/z_stepper_align.cpp#L64

qwewer0, 2020-08-09:
The on the fly offset edit did not come to my mind. Thanks for info.

closed 2020-08-09 by qwewer0