MarlinFirmware/Marlin #18929: steps per mm vs probe offset

Thijsvanzwam, 2020-08-05:
### Bug Description

Not sure if it a bug or I am misunderstanding it but I finetuned my steps per mm today and had to completely remeasure the probe offset.
In my opinion the probe offset is in mm and does not change if steppingstones per mm is changed, however my print did not even touch the bed on the first layer and after changing the offset to a different value, all was well.

I measured the probe offset the same way I did before and nothing has changed on the printer, except the steps per mm

### My Configurations

[Archief.zip](https://github.com/MarlinFirmware/Marlin/files/5031278/Archief.zip)

### Steps to Reproduce

1. recalculate the steps per mm
2. change them in the firmware and flash the printer
3. remeasure the probe offset

**Expected behavior:** no change in probe offset value

**Actual behavior:** big difference, went from -2.96 to -3.45

#### Additional Information

Does marlin translate the probe offset to steps or does it use mm?

ellensp, 2020-08-05:
You have EEPROM_SETTINGS enabled, eeprom variables always takes precedence over Configuration.h changes.  steps/mm and probe offsets are stored in eeprom.  Editing the firmware will not change these settings until eeprom is reset.

Thijsvanzwam, 2020-08-06:
After flashing i always do a m502 and m500. Should Thatcher nog Reset the eeprom? Even then i had to redo the probe offset

Thijsvanzwam, 2020-08-06:
But it is weird right? I measured the offset using a caliper. That setting always worked fine until the steps per mm were changed. I now use a setting that does not comply with the measurement done with a caliper.

ellensp, 2020-08-06:
which means the new steps/mm value you entered isnt correct.  You have effectively created you own unit of measurement.

Thijsvanzwam, 2020-08-06:
The previous steps per mm resulted in wrong prints, they were way of in measurement. These settings now make the print almost perfect. 
If I move the z axes up now and measure with a caliper it is correct as well. Only thing now wrong is the probe offset when measured with the caliper.

AnHardt, 2020-08-06:
Are you (still) aware, for most probes it is difficult to measure nozzle-probe-offset with a caliper? You have to measure to the probes trigger point - not its housing.

Thijsvanzwam, 2020-08-06:
I know, it is the Triggerpoint I measured with the caliper. It always printed perfectly!

Some strange update: 
Yesterday I finetuned the steps per mm, remeasured the probe offset as the filament didn’t even touch the buildplate and did some printing, all went well with the new settings.
Today I checked and the new probe offset that worked yesterday now grinds the nozzle in the bed. Changed back to the “old” setting (the caliper one) and printing is perfect again!

sjasonsmith, 2020-08-09:
I find it pretty hard to measure with a caliper down by my nozzle. I can get a rough estimate, but not less than 0.5mm accuracy. There is just too much stuff in the way.

Right now this does not seem like it is likely to be a Marlin bug. Please try the following to help us understand further details.

1. Enable `Z_MIN_PROBE_REPEATABILITY_TEST` and perform many probes using `M48`. We want to see the reported "range" of the probes. I suggest using `M48 P50` to perform the maximum 50 probes. If the range is large, then Z-offset will seem to change all the time.

2. Measure the Z offset by jogging the nozzle to the bed, not by using a caliper. This will be more accurate than trying to fit a measuring device inside the printer.

Thijsvanzwam, 2020-08-10:
the measurement using the caliper was done in a special mount, using a caliper normally used for measuring cars engine parts. accuracy to 0.001 mm. We did that when we rebuild the printer with the carriage apart. (the mounting had a steel plate simulair as the printed and I measured it from the triggerpiont)
I entered the caliper measurements in  the printer and the first print was spot on, all prints after that are too. When testing the settings by jogging the nozzle, they correspondent to the caliper measurement.

The problem is that when I update the firmware's steps per mm setting, the probe offset is corrupted. 
The offset is in mm and the distance does not change. When I update the steps per mm, the offset value should not be affected by that, the firmware just has to calculate the new amount of steps to reach the same distance as it dit before.

The steps I did:
- update the firmware with the new steps per mm settings. (just tiny difference to the previous setting like going from 99,8 steps to 100,10)
- perform M502 and M500
- put in the settings like probe offset and linear advance (both using the LCD, in the firmware I put them at 0 because I want to be able to change that on the LCD) and safe the settings to the eeprom.
- do a testprint and find out that the probe offset is suddenly wrong. 
- change the offset by jogging the nozzle to a distance that is about 1 mm different then the original settings
- do a new testprint which is fine
- turn off the printer and turn it back on the next day.
- print the same testprint but now the nozzle grinds in the bed. 
- put the "old" nozzle offset back using the LCD and all is well again.

Both the probe and the nozzle are mounted firmly and to me it looks like the eeprom is not updated correctly as turning off the printer seems to make a difference.

AnHardt, 2020-08-10:
> like going from 99,8 steps to 100,10)

Be sure to use "."-notation in your config file - not ",", for floats. Else it will make a difference of ~1%, not the intended 0.3% .

Thijsvanzwam, 2020-08-14:
I know, thanks.
The notations are correct in the firmware

Thijsvanzwam, 2020-08-16:
did the M48 P50.
results are:

mean: 0.001622
min: -0.004
max 0.007
range: 0.011

sjasonsmith, 2020-09-01:
Let me test my understanding:

You encounter problems if you adjust steps/mm without power-cycling? If you power-cycle after storing new steps/mm, the original offset works fine?

Thijsvanzwam, 2020-09-01:
When i adjust the steps/mm and do M502, M500 and even press the reset button after that the probe offset is wrong. even after a (short) power cycle.
I then readjust the offset to a larger setting (go from -2.96 to -3.40) and all is ok again. 
Turn off the printer overnight, turn it back on the next day and the nozzle just floats in the air. put the original setting back (-2.96) and it works again.

closed 2020-10-09 by github-actions[bot]