MarlinFirmware/Marlin #18931: Wrong result for value calculated by Marlin

discip, 2020-08-06:
### Bug Description

Value calculated by Marlin is wrong.

### My Configurations

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/5031739/Configurations.zip)

### Steps to Reproduce

I tried to make it a bit easier to change the amount of *microsteps* (***Configuration_adv.h***) by using some kind of formula to calculate the amount of *steps per unit*.
Because you would have to edit both ***Configuration.h*** and ***Configuration_adv.h*** every single time.
By doing so, I noticed, that the value for **`E0`** was calculated wrong.

1. In ***Configuration.h*** change line **751** to something like this:
**`#define DEFAULT_AXIS_STEPS_PER_UNIT   { (80.121*X_MICROSTEPS/16), (80.121*Y_MICROSTEPS/16), (399.778*Z_MICROSTEPS/16), (409*E0_MICROSTEPS/16) }`**
2. Compile and upload to your machine.
3. Check the value for **`E0`**

**Expected behavior:**
Value for **`E0`** should result in **`102.25 steps/mm`**.

**Actual behavior:**
Value for **`E0`** results in **`102.00 steps/mm`**.

### Interestingly only the *`E0`* value is affected.

sjasonsmith, 2020-08-06:
Try changing 409 to 409.0. Without the decimal anywhere in the formula you are performing integer arithmetic.

ellensp, 2020-08-06:
Confirmed, needs the .0     Not a marlin issue.

closed 2020-08-06 by ellensp

discip, 2020-08-06:
@sjasonsmith & @ellensp
## Thanks a lot! 👍