MarlinFirmware/Marlin #18932: [BUG] menu_cancelobject.cpp, SUBMENU_N supports only numbers up to 9

swissnorp, 2020-08-06:
### Bug Description

There is a problem in the way integers above 9 are added to menu items, it supports only 1 digit. [lcdprint.cpp](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/lcdprint.cpp)

Its a problem with SUBMENU_N in [menu_cancelobject.cpp](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/menu/menu_cancelobject.cpp) because only objects from 0-9 are displayed properly.
Items over 9 are displayed in special characters as a single digit.

The confirmation screen is allright.

### Steps to Reproduce

1. Activate `CANCEL_OBJECTS` in [Configuration_adv.h](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration_adv.h)
2. Print the test file from the zip: [cancel_object_more_than_10-test.zip](https://github.com/MarlinFirmware/Marlin/files/5034135/cancel_object_more_than_10-test.zip)
3. Go to menu CANCEL OBJECT
4. Scroll down to 9 and further

boelle, 2020-08-06:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

swissnorp, 2020-08-06:
I dont think they have anything to do with the issues but if you want them here theye come...
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/5034469/Configurations.zip)

ellensp, 2020-08-07:
For people like myself who have never used this menu and can't find it!  It is in Configuration|Advanced Settings|Cancel Object

swissnorp, 2020-08-07:
> 
> 
> For people like myself who have never used this menu and can't find it! It is in Configuration|Advanced Settings|Cancel Object

In current bugfix its below TUNE in the MAINMENU while printing! https://github.com/MarlinFirmware/Marlin/commit/6da477e6bf675c10e1eb5c3279a06dffeb0c3947

ellensp, 2020-08-07:
yip, my bugfix was 9 days old...  and someone moved it in that time...

thinkyhead, 2020-08-11:
Please test #18998 which should fix the issue.

swissnorp, 2020-08-13:
https://github.com/MarlinFirmware/Marlin/pull/18998 fixes this problem... closed

PR #18998 (2020-08-12): String interpolation for 2 digit numbers.
String index interpolation only supports 1 digit. This PR extends this to 2 digits and fixes the TFT implementation of the same method so that it can display "Chamber" and "Bed" strings.

Closes #18932
closed 2020-08-13 by swissnorp