MarlinFirmware/Marlin #18933: Z heigth after homing

lewiss66, 2020-08-06:
For info I use only manual nobbe adjustment for the bed. My concern here is I just want the nozzle to be raised at a certain height above the bed after homing that I set in Marlin firmware (2.0) . ex define Z_offset.. or Z_AFTER_HOMING but whatever I try the nozzle height never goes over the bed level after homing

GMagician, 2020-08-06:
### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

GMagician, 2020-08-06:
Z_AFTER_HOMING works only if Z_HOME_DIR < 0 but without cnf...

boelle, 2020-08-06:
@lewiss66 did you read the template???

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

ellensp, 2020-08-07:
Please attach your configuration files.

qr-buell, 2020-08-07:
i have the sam issue. I use MESH_BED_LEVELING. 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5041650/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5041651/Configuration_adv.zip)

boelle, 2020-08-08:
closing as @lewiss66 did not provide the configs that the template requires

we can reopen when they are attached

ellensp, 2020-08-09:
Configs are are attached to a reply...

closed 2020-08-09 by ellensp