MarlinFirmware/Marlin #18936: MKS Gen_L V2.0 Board - FIL_RUNOUT_PIN missing in Marlin 2.0 Bug Fix

hanermad, 2020-08-06:
FIL_RUNOUT_PIN not defined in pins_MKS_GEN_L_V2.h

defaults to 4 in pins_RAMPS.h - results in inop filament sensor

Resolved by adding matching definition for SERVO3_PIN

//*****From pins_Ramps.h
#ifndef FIL_RUNOUT_PIN
  #define FIL_RUNOUT_PIN                      32  //***** Defaults 4 in pins_Ramps.h
#endif

borland1, 2020-08-06:
That is normal.  You can just move your servo header wire to Servo Signal post D4.  File pins_MKS_GEN_L_V2.h includes statement: `#include "pins_RAMPS.h"`.  
If you want to override the pins file pin definition, then you can place this single code line in your `Configuration.h` file:

`#define FIL_RUNOUT_PIN 32`

No need to change the pins file.

thisiskeithb, 2020-08-06:
Since the GEN-L V2 doesn't explicitly call out a pin for for filament runout ([link to board pins diagram](https://github.com/makerbase-mks/MKS-GEN_L/blob/master/hardware/MKS%20Gen_L%20V2.1_001/MKS%20GEN_L%20V2.1_001%20PIN.pdf)), you're free to define one in `Configuration.h` as @borland1 pointed out.

closed 2020-08-06 by thisiskeithb

hanermad, 2020-08-07:
But your configuration.h file says it specifically uses SERVO3 for runout and it needs both for the logic to work. HIGH/LOW and PULLUP/PULLDOWN are defaulting to pin 4 in the pins_RAMPS.h
At least point people in the right direction in the comments!

/**
 * Filament Runout Sensors
 * Mechanical or opto endstops are used to check for the presence of filament.
 *
 * RAMPS-based boards use SERVO3_PIN for the first runout sensor.
 * For other boards you may need to define FIL_RUNOUT_PIN, FIL_RUNOUT2_PIN, etc.
 */
#define FILAMENT_RUNOUT_SENSOR  //*****
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define NUM_RUNOUT_SENSORS   1     // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_STATE     LOW   // Pin state indicating that filament is NOT present.  //***** Was LOW
  #define FIL_RUNOUT_PULLUP          // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN      // Use internal pulldown for filament runout pins.

hanermad, 2020-08-07:
I know....after a lot of trial and error. I put it in my pins file since it's defined in the pins_ramps.h for the default.

The comments are a little confusing since the MKS GEN_L V1 & V2 are both Ramps boards. I've seen a lot of confusion will searching for a solution.

hanermad, 2020-08-07:
Would this be an acceptable change?

/**
 * Filament Runout Sensors
 * Mechanical or opto endstops are used to check for the presence of filament.
 *
 * RAMPS-based boards use SERVO3_PIN for the first runout sensor.
 * For other boards you may need to define FIL_RUNOUT_PIN, FIL_RUNOUT2_PIN, etc.
 */

#define FILAMENT_RUNOUT_SENSOR  //*****
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define NUM_RUNOUT_SENSORS   1     // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_PIN       32    //***** Also req'd for Ramps boards where SERVO3 does not match FIL_RUNOUT_PIN in pins_RAMPS.h, needs to match for logic to work
  #define FIL_RUNOUT_STATE     LOW   // Pin state indicating that filament is NOT present.  //***** Was LOW
  #define FIL_RUNOUT_PULLUP          // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN      // Use internal pulldown for filament runout pins.