MarlinFirmware/Marlin #18943: M600 layer gap

Ratler5, 2020-08-07:
I apologize in advance I'm not sure what i need to put here to help track this down let me know and i will get it! 

Every time i do a M600 when the print resumes there is a gap in the layers. 
just compiled the firmware with latest bug fix branch and config example for 
Ender 3, SKR mini E3 2.0

Expected behavior
M600 is issued from terminal/gcode/filament run-out sensor, printer parks, passes, eject filament and waits for user 
User replaced filament and click resume. 
print picks up where it left off

observed behavior 
Print resumes, but is about 2/3 a layer higher than when M600 triggered. So it does continue but there is a viable gap and breaks easily 

[image]

boelle, 2020-08-07:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Ratler5, 2020-08-07:
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/5043042/Archive.zip)

thanks :)

ellensp, 2020-08-07:
This could be backlash in your z hardware.   (not saying that it is, could be) 
This is often seen when you move to a location from a different direction.
Ie the first to nth layers are always upwards moves. So the axis height was set with lower thread of the leadscrew.
But then you park and the heads moves up then down to same position. This layer height is set with upper part of the leadscrew.
If you have a dial gauge you could mount it on your printer and test it. reading values directly from the gauge.

boelle, 2020-08-08:
or it could be a to loose idler in the extruder, happend to me on an gen. e3d titan where the idler screw had backed all the way out

to tight can also be a problem

Ratler5, 2020-08-12:
I did a full rework on my x gantry and tightened everything and it solved the issue. not sure why it decided to do this right after i changed the board... talk about timing! 

I will close this now

closed 2020-08-12 by Ratler5