MarlinFirmware/Marlin #18945: [BUG] (Error when build for laser control)

annhan, 2020-08-07:
Hi, I haven't built my firmware with laser control. I am using the ULTRATRONICS board.
My error:: error: "'set_pwm_duty' was not declared in this scope
       set_pwm_duty(pin_t(SPINDLE_LASER_PWM_PIN), ocr ^ SPINDLE_LASER_PWM_OFF);
       ^~~~~~~~~~~~"
i have checked the set_pwm_dutty function just support for the LPC1768 device. Am I right ?
and how can i fix this ?
thank you

ellensp, 2020-08-07:
comment out SPINDLE_LASER_FREQUENCY   this is only for 8 bit avr and lpc1768

ellensp, 2020-08-07:
Or manually apply the above PR

annhan, 2020-08-07:
OK, I don't see a comment line (  Spindle/laser frequency (only on supported HALs: AVR and LPC). This was enabled at default. thank you so much

ellensp, 2020-08-07:
Pr has been added to bugfix so future users wont fall into 'trap'

closed 2020-08-07 by ellensp