MarlinFirmware/Marlin #18953: [FR] HOTEND_IDLE_TIMEOUT - Gcode way to disable or other options

jr-air, 2020-08-08:
Description: 
With HOTEND_IDLE_TIMEOUT enabled, and a pre-heated hotend, print jobs can fail. Appears to be primarily due to the time it takes the printbed to heat up and/or time for ABL to finish

Feature Request:
Either a gcode which can be added to the print startup script or other methods for disabling the timer which may just be more than monitoring for extruder movement. (maybe monitor for movement on all axis?)

To Replicate:
1) Pre-heat hotend above 180c  (or value above HOTEND_IDLE_MIN_TRIGGER)
2) submit print job with bed not pre-heated and with a temp which takes longer than the HOTEND_IDLE_TIMEOUT_SEC value.
2a) also with ABL, more time is taken before first extruder movement

Clearly not a bug and the workaround is to reduce the hotend temp below HOTEND_IDLE_MIN_TRIGGER value, waiting for this lower temp to be reached, then starting printjob.

(HOTEND_IDLE_TIMEOUT is a great addition otherwise! Thanks.)

thisiskeithb, 2020-08-08:
If your bed heat up or ABL routine takes an abnormally long time, increase `HOTEND_IDLE_TIMEOUT_SEC` to something more than than the five minute default.

jr-air, 2020-08-09:
> If your bed heat up or ABL routine takes an abnormally long time, increase `HOTEND_IDLE_TIMEOUT_SEC` to something more than than the five minute default.

Yes, this is another workaround or option I considered, though I change between PLA/PETG/ABS filament often, and would have to "hard code" a very high value to cover all cases. As this is still a great, much needed, addition to Marlin which I will continue to use, can't say if I can get used to "hard coded" timers. :) 

Though since I use Octoprint for all my printers, I came up with a much easier workaround. There is an option for "gcode before print job starts". I just threw in gcode to drop and wait for the hotend temp goes below  "HOTEND_IDLE_MIN_TRIGGER". (Yes there is also Octoprint plugin for hotend temp timeout, but having this timer in firmware is much much better!)

Thanks.

thisiskeithb, 2021-09-24:
`HOTEND_IDLE_TIMEOUT` no longer times out if Marlin is busy: https://github.com/MarlinFirmware/Marlin/pull/21535

closed 2021-09-24 by thisiskeithb