MarlinFirmware/Marlin #18954: [BUG] Endstops won't stay triggered on homing XY

qwewer0, 2020-08-08:
### Bug Description

Before commit 9ee891c, Endstop States would stay triggered, but now it stays open.

### My Configurations

SKR Mini E3, Ender 3, Latest Bugfix-2.0.x
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5045156/Configuration.zip)

### Steps to Reproduce

1. Home with `G28 X Y`
2. Check `Endstop States` with `M119`

**Expected behavior:** When homed XY we should get `x_min: TRIGGERED`, and `y_min: TRIGGERED`

**Actual behavior:** When homed XY we get `x_min: open`, and `y_min: open`

GMagician, 2020-08-08:
I think it's because of `TMC_HOME_PHASE { 640, 640, -1 }`. With it you are saying to align to stepper phase. And selected phase is far from endstop (of course it can't be the one against).
Solution is to disable TMC_HOME_PHASE.

Previous code was buggy

qwewer0, 2020-08-08:
@GMagician Changed it to `TMC_HOME_PHASE { 384, 384, -1 }`, witch wasn't "working" before 9ee891c and now it is working and the endstops stays triggered. Thank you!

closed 2020-08-08 by qwewer0