MarlinFirmware/Marlin #18955: Should 'ok' after resend depend on command queue?

mbenkmann, 2020-08-08:
https://github.com/MarlinFirmware/Marlin/blob/fd9c6983d30f70171d9d9a9c55962689fa4cdd85/Marlin/src/gcode/queue.cpp#L311

After a Resend request, Marlin calls ok_to_send().
ok_to_send() sends an 'ok' if and only if the next command in the command queue has its send_ok[]==true.
I do not see how the next entry in the command queue has any relation to the Resend request. I'm pretty sure we always want to send an 'ok'.
Additionally, what if the command queue is empty and an incorrect line number comes in? The resend request would call ok_to_send() and it would dereference send_ok[cmd_queue_index_r] as well (if ADVANCED_OK) as command_queue[cmd_queue_index_r] despite the fact that both entries are invalid.

boelle, 2020-08-08:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

mbenkmann, 2020-08-08:
THIS IS A DEVELOPMENT RELATED ISSUE. IT'S ABOUT THE FUCKIN' CODE.

boelle, 2020-08-08:
the issue list is for bugs only and FR's

mbenkmann, 2020-08-08:
I am reporting a bug.

mbenkmann, 2020-08-08:
The referenced line is a bug. Please pass this on to someone who understands the code.

boelle, 2020-08-08:
you forgot to read the template, and i only see question marks

and the title did not say [BUG]

SO HOW THE FUCK AM I SUPPOSED TO KNOW?

mbenkmann, 2020-08-08:
BY READING

mbenkmann, 2020-08-08:
I used github's comment on code function. it does not open a template.

mbenkmann, 2020-08-08:
How many users reference code lines? How many users use words like "dereferences"? How many users say stuff like "command queue is empty "

mbenkmann, 2020-08-08:
I'm trying to help you fix your code and you're treating me like some moron who wants to know how to plug in an SD card.

boelle, 2020-08-08:
> How many users reference code lines?

first time i see it in a VERY long time

boelle, 2020-08-08:
> your code

not  my code

mbenkmann, 2020-08-08:
If you're not considering yourself part of the Marlin project, give the bug to someone who does.

boelle, 2020-08-08:
> like some moron

i would not call you that, more like a person that could need help with his choice of words

boelle, 2020-08-08:
> give the bug to someone who does.

i have not taken the bug

mbenkmann, 2020-08-08:
According to git blame, @thinkyhead touched this code the last. Maybe he should take a look.

boelle, 2020-08-08:
> @thinkyhead touched this code the last.

he is also the über boss arround here, everyone else is just "helpers"

sjasonsmith, 2020-08-09:
Do you think you could come up with a way to reproduce a failure, to prove this can happen as you suspect?
If so, then I suggest fixing it and posting a pull request with the change.

mbenkmann, 2020-08-09:
(NOTE: I do not have a printer running Marlin 2.x, but since I spotted the issue in the current code I expect it to be reproducible in the same way.)

I was able to reproduce the issue like this:

1. Log into my Raspberry Pi on the shell
2. Make sure Octoprint is NOT connected to the printer
3. Talk to the printer on the shell

       socat STDIO /dev/ttyUSB0,b115200,cfmakeraw

4. Initialize the line number counter
   
       N1M110N1*125

5. Start a print from the SD card via the printer's physical user interface interface (i.e. not via Octoprint etc.)
6. While the print is running (in my case during heatup), send a command with an incorrect line number

       N5G28*54
    
7. This is my result

```
T:200.1 E:0 W:8
T:200.0 E:0 W:7
T:200.2 E:0 W:6
N5G28*54
T:200.0 E:0 W:5
T:200.0 E:0 W:4
T:200.0 E:0 W:3
T:200.0 E:0 W:2
T:200.0 E:0 W:1
T:200.0 E:0 W:0
Error:Line Number is not Last Line Number+1, Last Line: 1
Resend: 2
Setting Print Acceleration: 500.00
Setting Retract Acceleration: 1000.00
Setting Travel Acceleration: 500.00
Done printing file
echo:0 hours 0 minutes
echo:enqueueing "M84 X Y Z E"
```

As you can see, the Resend request was not followed by an 'ok', presumably because the send_ok[] array contained the value "false" from the SD card commands which are committed with the value false.

The fix that takes the least amount of bytes in the firmware is probably to insert the line

        send_ok[index_r] = true;

right before the call to ok_to_send() in GCodeQueue::flush_and_request_resend()

mbenkmann, 2020-09-09:
Well, what about my suggestion to just insert  

send_ok[index_r] = true;

AnHardt, 2020-09-09:
The question, for me, is: Has the `Error:Line Number is not Last Line Number+1, Last Line: 1` to be followed by an `ok`. It's definitely not 'ok' - else we would not send an error. The hosts commandqueue has to be reset anyway.

mbenkmann, 2020-09-09:
Well, Marlin currently sends an ok after a Resend in all but the most obscure cases. You can assume every program that interfaces with Marlin like Octoprint to expect an ok. You'd be changing the behavior by removing the ok. This bug report is about an obscure situation where no ok is sent. It is not the place to discuss a fundamental change in Marlin's protocol.

mbenkmann, 2020-09-09:
In other words: To fix the bug, insert 

send_ok[index_r] = true;

and then discuss whether that should be changed to "false" in some future version of Marlin.

closed 2020-10-18 by github-actions[bot]