MarlinFirmware/Marlin #18956: [BUG] IDEX Duplication and G29 issue.

Romain3481, 2020-08-08:
### Bug Description

Hi !
I put M605 S2 T0 X110 on the start G-code of my slicer, but it doesn't work well :
- With G29 on my start G-code, E0 heats during autolevel, but not E1... the job starts when E0 reached its temperature, and E1 starts to heat when the job starts !!!! E1 is too cold and of course the E1 extrusion doesn't work.
- Without G29 in the start G-code, both extruders heat at the same time, but the job starts without autolevel :(
What'w wrong with M605 S2 and G29 ?
Thank you ! :)

### My Configurations
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5045526/config.zip)
Marlin 2.0.5.4
BTT GTR 1.0

sjasonsmith, 2020-08-09:
Please test with version 2.0.6 to see if this is still an issue. You could also test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip), although it isn't much different than 2.0.6 at this point.

Romain3481, 2020-08-10:
Hi,
I solved the issue,
I had to put M104 S[first_layer_temperature] just after M605 S2 X110 R0 in the start G-code.

closed 2020-08-10 by Romain3481