MarlinFirmware/Marlin #18957: [FR] Increase PWM frequency for 32-bit platforms

trengtor, 2020-08-08:
The used frequency of 500 Hz often creates a distinctly audible sound on the fan windings. Is it possible to add an option to increase this frequency so that, if necessary, it would be possible to shift such noise into an inaudible area? Soft-PWM does not solve this problem, as it either creates a rumble, or dramatically reduces the control range.

sjasonsmith, 2020-08-08:
The wide variety of 32-bit hardware makes it a challenge to offer this consistently. Can you post more information about the hardware you are using, along with your configuration files?

trengtor, 2020-08-08:
> The wide variety of 32-bit hardware makes it a challenge to offer this consistently. Can you post more information about the hardware you are using, along with your configuration files?

STM32F103RC_btt_512K (SKR mini E3 V1.2). Currently using Soft-PWM.

sjasonsmith, 2020-11-12:
Both STM32F1 and STM32 HALs now support fast PWM. You can enable this for fans using `FAST_PWM_FAN` and adjust the frequency with by setting `FAST_PWM_FAN_FREQUENCY`. The default frequencies are likely too fast for your use.

Is  there a specific "32-bit platform" you are interested in? There is enough variety in them now that it is unlikely they would ever all be implemented at the same time. I don't even know which other HALs do or do nor support it.

trengtor, 2020-12-27:
> Both STM32F1 and STM32 HALs now support fast PWM. You can enable this for fans using `FAST_PWM_FAN` and adjust the frequency with by setting `FAST_PWM_FAN_FREQUENCY`. The default frequencies are likely too fast for your use.
> 

Ok, thanks! I will try to use this.

trengtor, 2020-12-27:
> Both STM32F1 and STM32 HALs now support fast PWM. You can enable this for fans using `FAST_PWM_FAN` and adjust the frequency with by setting `FAST_PWM_FAN_FREQUENCY`. The default frequencies are likely too fast for your use.

trengtor, 2020-12-27:
> Both STM32F1 and STM32 HALs now support fast PWM. You can enable this for fans using `FAST_PWM_FAN` and adjust the frequency with by setting `FAST_PWM_FAN_FREQUENCY`. The default frequencies are likely too fast for your use.

I tryed to complile but has error:
#error "Features requiring Hardware PWM (FAST_PWM_FAN, SPINDLE_LASER_FREQUENCY) are not yet supported on STM32F1."
[image]

ellensp, 2020-12-27:
@trengtor This feature is only in bugfix branch of marlin at this time. It id most definitely not in an old 2.0.6.1 marlin

trengtor, 2020-12-27:
> @trengtor This feature is only in bugfix branch of marlin at this time. It id most definitely not in an old 2.0.6.1 marlin

Ok, thanks

freakezoid1604, 2020-12-31:
is it possible to have this function for a RADDS/DUE combo also? the noice of the cooling fan is horrible. i also try the softpwm but...

Take a look to the video and hear the sound. Fan at 19% Speed at 12v
Fantype:  Sunon Radial Lüfter 50x50x15mm MF50151V1-A99 DC 12V 5000 U/min 40dBA Vapolager 2 Litzen 

https://user-images.githubusercontent.com/28877844/103415117-129bd900-4b81-11eb-8c52-d5e0cf37cf23.mp4

qwewer0, 2021-01-01:
This can be closed, as it is already implemented (bugfix-2.0.x)

sjasonsmith, 2021-01-01:
> This can be closed, as it is already implemented (bugfix-2.0.x)

It is only implemented for some platforms, but I am going to close it anyway.

The implementation of this is very specific to each family of hardware and their corresponding HAL. A broad request like this covering all 32-bit hardware has little hope of ever being complete.

closed 2021-01-01 by sjasonsmith

namadori, 2021-01-18:
> @trengtor This feature is only in bugfix branch of marlin at this time. It id most definitely not in an old 2.0.6.1 marlin

Hello, can you give us a reference so we can check when this feature is merged into mail 2.0.x branch? Thank you.

qwewer0, 2021-01-19:
> Hello, can you give us a reference so we can check when this feature is merged into mail 2.0.x branch? Thank you.

When the [new version](https://github.com/MarlinFirmware/Marlin/releases) is out.