MarlinFirmware/Marlin #18959: [BUG] Reworked build for PlatformIO fails to find compiler in custom installations

Creat, 2020-08-08:
### Bug Description

Since the rework of the PlatformIO build system I can no longer build Marlin at all. Neither the current 2.0.x official release, nor bugfix-2.0.x. I instantly get a "compiler not found" error. This also happens when I download a fresh and unedited copy of marlin and try to just compile it from VS Code:
```
Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1768; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
("Couldn't find a compiler! Fallback to", 'CC')
CC -D__MARLIN_FIRMWARE__ -DU8G_HAL_LINKS -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h
'CC' is not recognized as an internal or external command, operable program or batch file.

```
The reason is I found (after HOURS of searching) is that PlatformIO is installed globally for me, not (just) as part of VS Code or Atom directly. The same installation is used by them and myself when compiling manually without IDE.
On my system this is done with by setting the environment variable PLATFORMIO_HOME_DIR (set to `C:\Development\PlatformIO`). If I then install PlatformIO IDE (in Atom or VS Code), this is the location for PlatformIO itself, and toolchains, libraries and so on. [Here is a reference in the documentation](https://docs.platformio.org/en/feature-platformio-30/envvars.html), essentially it's a global override for the [`home_dir`](https://docs.platformio.org/en/feature-platformio-30/projectconf.html#projectconf-pio-home-dir).

This would technically could still work fine, if I had happened to install it into a folder called `platformio` (note the capitalization), which is the default folder name (usually located in %HOMEPATH% or ~ for unix based systems).
It tracks back to the regex that is used to find the compiler in common-dependencies.py (declared [here](https://github.com/MarlinFirmware/Marlin/blob/fd9c6983d30f70171d9d9a9c55962689fa4cdd85/buildroot/share/PlatformIO/scripts/common-dependencies.py#L142), used [here](https://github.com/MarlinFirmware/Marlin/blob/fd9c6983d30f70171d9d9a9c55962689fa4cdd85/buildroot/share/PlatformIO/scripts/common-dependencies.py#L151)). It assumes and/or requires the PlatformIO installation folder name ends an all lower case "platformio". So anyone having it installed in a differently named folder (like `C:\MyFavoriteTool`) can no longer compile Marlin.

Note that this isn't limited to windows, I just happen to be on windows. As far as I know this works on any OS supported by PlatformIO

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
I have attached my config since it's understandably required, but as I already wrote above, it's irrelevant since it's about not finding the compiler, before the configs are even read (by the compiler, that isn't found).

[Marlin-Config-Creat.zip](https://github.com/MarlinFirmware/Marlin/files/5046183/Marlin-Config-Creat.zip)

### Steps to Reproduce

1. Uninstall/remove PlatformIO from VS Code if both are installed, or install VS code if it isn't
2. Exit VS Code if running
3. Create an environment variable PLATFORMIO_HOME_DIR with a path for it to install to, not ending in lowercase "platformio" (and with write permission, obviously)
4. Checkout a fresh copy of Marlin, or delete the .pio in an existing copy (otherwise the compilerpath is cached here).
5. Start VS Code and (re)install PlatformIO IDE
6. Open and try to compile the copy of Marlin from step 4.

As far as I know, this _should_ reproduce it, but I haven't actually tried (been at this for like 5 hours how already)

**Expected behavior:** The compilers compiles marlin

**Actual behavior:** can't even find the properly installed compiler

#### Additional Information

* Obviously I'll do my best to provide any additional information that might be needed (or test something).

rhapsodyv, 2020-08-08:
Do this branch solves the problem for you?

https://github.com/rhapsodyv/Marlin/tree/fix-18959

Creat, 2020-08-08:
Yes and no. It does fix it in my current case, but not in general.
If someone has it installed in `c:\devstuff\platformio_public` of even `c:\devstuff\platform io` it still wouldn't work. I just happen to have it installed in a folder name that only differs in capitalization. Others might not.

But yes, for me personally in my current installation this would of course solve the problem.

rhapsodyv, 2020-08-08:
> ally in my current installation this would of cours

Perfect. Sure, we can solve all corner cases :-)

Thanks!

rhapsodyv, 2020-08-08:
Just sent the fix for non default install folders.

thisiskeithb, 2020-08-09:
Does https://github.com/MarlinFirmware/Marlin/pull/18960 resolve this issue?

Creat, 2020-08-09:
Yes, the correct path is filled in now.

I have another issue that I had assumed is related to the compiler not being found, but it isn't resolved:
All of the macros used in `#if` like `ENABLED(...)` and so on confuse the IDE (but it compiles fine). They all result in `function call is not allowed in a constant expression`. That was also not a problem until I had pulled/updated a few days ago. This leads to all sections inside any `#if ... #endif` to not be shown as enabled/disabled, making finding relevant section much harder.

Example view here. Note all the red on the right.
[image]

thisiskeithb, 2020-08-09:
I get that too. The linter has trouble keeping up with Marlin’s macros.

As long as it builds fine, you’re good to go 🙂

closed 2020-08-09 by thisiskeithb

Creat, 2020-08-15:
Got an update on the "intellisense" errors and how to fix them:

First of all, the (main?) problem is that by default the interpreter/linter doesn't recognize named variables in variadic macros (as it's used in the ENABLED-macro, which as far as I know is a C++ extension to macros. So the C/C++ extension just needs to be told what language standard to use.

If you go to the C/C++ section in the VS Code settings (under extensions) there's an option "C_Cpp: Intellisense mode" and should (probably?) be set to gcc-arm. Settings-ID: C_Cpp.default.intelliSenseMode

I don't think this can be added as a default to the repository, since the setting doesn't support the folder-scope, but maybe it might be worth mentioning this in the VS Code installation/usage guide?

Another option might be to add a workspace-file to the repository (same place as the platformio.ini I suppose), since it can be saved there. But I don't know how that interacts with opening the folder and such (or if it has to be explicitly opened as a workspace then). The file contents would essentially just be a couple of lines tho.

PR #18960 (2020-08-08): Fix compiler search in non default PIO installs -  Fix #18959 
### Description

Fix the search for the current gcc for non standard PIO installs

### Benefits

Less build broken
Fix #18959 

### Related Issues

#18959