MarlinFirmware/Marlin #18961: SKR PRO MARlIN 2.06 error: 'HAL_TIMER_RATE' was not declared in this scope; did you mean 'PULSE_TIMER_RATE'?

devilcorp, 2020-08-08:
[image]

[config files.zip](https://github.com/MarlinFirmware/Marlin/files/5046521/config.files.zip)

hardware:
Anycubic predator
SKR pro 1.1
TMC2209

Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/pinmap.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/rtc.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/spi_com.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/startup_stm32yyxx.S.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/stm32_def.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/stm32_eeprom.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/system_stm32yyxx.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/timer.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/twi.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/uart.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/usb/cdc/cdc_queue.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/usb/cdc/usbd_cdc.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/usb/cdc/usbd_cdc_if.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/usb/hid/usbd_hid_composite.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/usb/hid/usbd_hid_composite_if.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/usb/usb_device_core.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/usb/usb_device_ctlreq.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/usb/usb_device_ioreq.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/usb/usbd_conf.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/usb/usbd_desc.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/usb/usbd_ep_conf.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/stm32/usb/usbd_if.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/syscalls.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/wiring_analog.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/wiring_digital.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/wiring_pulse.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/wiring_shift.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/FrameworkArduino/wiring_time.c.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/HAL.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/HAL_SPI.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/MarlinSerial.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/Sd2Card_sdio_stm32duino.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/Servo.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/SoftwareSerial.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/eeprom_flash.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/eeprom_sdcard.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/eeprom_sram.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/eeprom_wired.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/fastio.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/timers.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/usb_serial.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/watchdog.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/HAL_spi_L6470.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/backtrace/backtrace.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/backtrace/unwarm.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/backtrace/unwarm_arm.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/backtrace/unwarm_thumb.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/backtrace/unwarmbytab.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/backtrace/unwarmmem.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/backtrace/unwinder.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/backtrace/unwmemaccess.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/eeprom_api.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/eeprom_if_i2c.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/eeprom_if_spi.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/esp_wifi.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/HAL/shared/servo.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/MarlinCore.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/core/multi_language.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/core/serial.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/core/utility.cpp.o
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/feature/babystep.cpp.o
[... 924 lines cut ...]
STM32F070RB_malyan        IGNORED
malyan_M300               IGNORED
chitu_f103                IGNORED
chitu_v5_gpio_init        IGNORED
STM32F103RET6_creality    IGNORED
STM32F401VE_STEVAL        IGNORED
FLYF407ZG                 IGNORED
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           FAILED    00:01:20.122
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
LERDGEX                   IGNORED
LERDGES                   IGNORED
LERDGEK                   IGNORED
rumba32                   IGNORED
esp32                     IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
linux_native              IGNORED
include_tree              IGNORED
=========================================================================== 1 failed, 0 succeeded in 00:01:20.122 ===========================================================================
The terminal process "platformio 'debug'" terminated with exit code: 1.
 
Terminal will be reused by tasks, press any key to close it.

ellensp, 2020-08-08:
HAL_TIMER_RATE is defined in timer.cpp after it is needed in timers.h

Edit Marlin/src/HAL/STM32/timers.h find  
```
// ------------------------
// Defines
// ------------------------
```
And add these lines 

```
#ifdef STM32F0xx
  #define MCU_TIMER_RATE (F_CPU)      // Frequency of timer peripherals
  #define MCU_STEP_TIMER 16
  #define MCU_TEMP_TIMER 17
#elif defined(STM32F1xx)
  #define MCU_TIMER_RATE (F_CPU)
  #define MCU_STEP_TIMER  4
  #define MCU_TEMP_TIMER  2
#elif defined(STM32F401xC) || defined(STM32F401xE)
  #define MCU_TIMER_RATE (F_CPU / 2)
  #define MCU_STEP_TIMER  9
  #define MCU_TEMP_TIMER 10
#elif defined(STM32F4xx) || defined(STM32F7xx)
  #define MCU_TIMER_RATE (F_CPU / 2)
  #define MCU_STEP_TIMER  6           // STM32F401 has no TIM6, TIM7, or TIM8
  #define MCU_TEMP_TIMER 14           // TIM7 is consumed by Software Serial if used.
#endif

#ifndef HAL_TIMER_RATE
  #define HAL_TIMER_RATE MCU_TIMER_RATE
#endif

```

This will compile, but I have no way to test it  (I don't have the hardware)

thisiskeithb, 2020-08-09:
Does your hardware work correctly with the patch?

sjasonsmith, 2020-08-09:
The reason those are done in the CPP instead of the header is that there is a complicated relationship between the various headers, and you could end up with one rate used in timers.cpp, and another used in the rest of Marlin.

That only impacts you if you are trying to override it from a pins file. If you are using the defaults it should be fine.

I haven't pulled the configs down yet to see what it encountering issues. I've been building SKR Pro firmware without encountering this. Looks like maybe the babystep options having issues...

devilcorp, 2020-08-09:
delta autocalibration ok, bed level ok, hot end heating ok, bed heating ok,
print pending, first layer ok, babystep ok

sjasonsmith, 2020-08-09:
Looking in stepper.cpp, the pulse widths for babystepping are handled a little bit differently than for normal step pulses.

They both use the timer, but babystepping bases the delay on cycles and normal steps base it on time. This prescaler variable is only needed for babystepping.

It would make sense to try to combine these, and do them the same way.

Moving stuff to the header is going to harm the ability to override the rate from a pins file, which is currently used by the Rumba32. If a mechanism can be found to obtain the timer's clock frequency from the framework, the need to specify it in a pins file could go away.

One thing to think about when addressing this is that even though the values _look like_ constants, they actually are not. `F_CPU` on STM32 is not known at compile time, and resolves into a function call. This means it might be reasonable to refactor this so that it isn't present in the header file at all, the macro would just call a function in the CPP file to get the rate.

This is used in operations which query the current count from the timer. This is an _incredibly slow_ operation, so the extra overhead of a function call or two isn't even going to be noticeable.

sjasonsmith, 2020-08-09:
I just found this:

```
/**
  * @brief  This function return the timer clock frequency.
  * @param  tim: timer instance
  * @retval frequency in Hz
  */
uint32_t HardwareTimer::getTimerClkFreq()
```

Integrating this in _should_ allow Marlin's timer code to use the proper timer rates without needing to specify them (or override them) ourselves. This is pretty long function call, so it would be best to do it once and cache the result.

devilcorp, 2020-08-09:
ok ok i see. 
your first patch works tell me what you want me to test and I do it

ellensp, 2020-08-09:
sjasonsmith is more typing out loud..   looking for a better way to fix it than my crude hack. 
I'm sure a PR will be forth coming in due time.

devilcorp, 2020-08-09:
;-) good job !

sjasonsmith, 2020-09-09:
Is that the STM32 framework is upgraded I can look into fixing this.

Deneteus, 2020-09-13:
I just got the SKR E3 DIP V1.1 and I got the following error with baby-stepping enabled. Compiling from bugfix. 

[image]
[e3 dip v1.1 babystepping error.txt](https://github.com/MarlinFirmware/Marlin/files/5213512/e3.dip.v1.1.babystepping.error.txt)

sjasonsmith, 2020-09-13:
@Deneteus that error has nothing to do with this issue.

arminth, 2020-10-21:
Today after updating my local copy of Marlin bugfix with the latest commits, I ran into the exact same issue. I also have a SKR1.1 Pro on an Anycubic Kossel Linear plus with TMC2208 steppers. I changed the timers.h file according to @ellensp patch and it compiled ok.
Strange thing is that 2.0.x-bugfix compiled fine on October-16 Version. So something must have changed in the meantime...

arminth, 2020-10-21:
I found what triggered the compile-error in my case. I had seen that the latest commits vonflicted with my stepperdriversettings in configuration.h. By accident, I forgot to change the E-stepperdriver from A4988 to TMC2208. As soon as there is a A4988 driver defined, the compiler error ist triggered. I corrected my setting to TMC2208 and then it compiled also without the patch.
Maybe this helps chasing the bug down.
Cheers
Armin

sjasonsmith, 2020-11-01:
The compile error regarding HAL_TIMER_RATE involved babystepping code paths. Any changes to babystepping options or default code paths could have hidden it.

I've posted a pull request which will resolve this and making modifying the timer header unnecessary.

PR #19978 (2020-11-01): Read STM32 clock rates from framework
### Description

STM32 timer clock rates are not always constant across a device. Depending on the timer they may be `F_CPU`, `F_CPU/2`, or maybe even other values (there are lots of STM32 devices, and I haven't read all their datasheets!).

This change reads the timer's clock rate from the framework, rather than assuming Marlin knows the correct value. This makes our code more generic and easier to implement across the wide range of devices that will be supported by the STM32 HAL.

This also removes the ability to override `HAL_TIMER_RATE`, which was adde [... cut]
closed 2020-11-01 by thisiskeithb