MarlinFirmware/Marlin #18964: First move after G28 sometimes incorrect

cbaugher, 2020-08-09:
### Bug Description
Sometimes the initial axis move immediately after homing is wrong, in that the wrong axis moves.
Found with the latest bugfix version on a Hictop D3 Hero printer (IDEX).

### My Configurations
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5046624/Configs.zip)
[Output.txt](https://github.com/MarlinFirmware/Marlin/files/5046642/Output.txt)

### Steps to Reproduce
This is a weird one.  To reproduce I do the following:

1) Connect to the printer with a serial monitor.
2) send G28, wait for completion. 
3) send G1Y20, wait for completion [A]
4) send G28, wait for completion
5) send G1X40, watch in confusion

What happens at step 5 is that there will be a brief pause, and then the Y axis will move at what seems to be the slow homing feed rate.  After it moves the same distance as it did in step 3, the Y axis will then return to where it started from (home) and the X axis will simultaneously move to the commanded position.

For additional weirdness, if you then send G1X80, the X axis will move to the new position but at a different feed rate from the previous X command.

[A] Depending on what you've done before step 2, the Y axis in step 3 may also move at what appears to be the slow homing feed rate instead of the commanded rate (if you specify one).  I've even seen it change feed rates in the middle of a single move!

#### Additional Information

It seems to have something to do with being configured for IDEX.  If I disable IDEX in the configuration and set it up for only a single extruder, the issue goes away.

IGHMC-Reg, 2020-08-09:
Hi CBaugher, I can confirm I get the same behaviour using your config and pins. I'm working on porting the tenlog screen and did my own Pins to support that and whilst I still occasionally get odd Y axis feedrates the problems aren't so bad. Happy to discuss/collaborate?

cbaugher, 2020-08-10:
> 
> 
> Hi CBaugher, I can confirm I get the same behaviour using your config and pins. I'm working on porting the tenlog screen and did my own Pins to support that and whilst I still occasionally get odd Y axis feedrates the problems aren't so bad. Happy to discuss/collaborate?

Hi,
I'm guessing you're Reg from the Facebook group?  Nice to see some other Hictop users here either way.

Thanks for the bug confirmation.  You're right in that it doesn't seem to be a total show stopper.  It still prints, at least in single extruder mode anyway.  The only reason I even noticed it at all is because after the normal probing routine it takes an absolute eternity for the T0 head to return back to the other side of the carriage.  So I guess that means it's not just after homing but also after a G29.  Not sure what these have in common but it's a place to start looking.

IGHMC-Reg, 2020-08-10:
Yes its me, hello! I wasn't seeing a lot of firmware info in the FB group and there don't seem to be many of us here either, possibly because support is so limited and maybe we can change that.  I don't use bed levelling so we can guess thats its not being *caused* by that and  I've only seen the slow movement in the Y. I bumped up max feed rates in config and I haven't seen it since.  I have a version that uses the Tenlog screen and will dry run two heads, as long as i can get through the homing sequence without them crashing. switching tools first and then homing seems to start the problems so I'm going to be looking at the 'workplace offsets' today. 80% of my problem is figuring out what the existing coding intent is/was without overlaying what we're trying to add!

cbaugher, 2020-08-12:
Yeah I suspect there's something that doesn't get reset after homing.  I'm also pretty sure it's related to IDEX, maybe not in the tool change routine specifically, but related in the sense that changing heads also triggers some state to get messed up.  I've gone through the toolchange code and so far haven't found anything that seems like it would cause this issue.  It is rather gnarly though so I'm also finding that understanding the intent is not easy.

And regarding support for the display, I opened an issue for discussion: #18999

cbaugher, 2020-09-03:
Just wanted to confirm that this issue still exists in the most current bugfix-2.0.x.

thinkyhead, 2020-10-04:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

cbaugher, 2020-10-05:
Confirmed with the latest bugfix-2.0.x branch.  The exact same behavior as before is observed.

C|

closed 2020-11-13 by github-actions[bot]