MarlinFirmware/Marlin #18966: NO SD CARD

diver197, 2020-08-09:
stm32f103re spi2 sd CARD, NO SD CARD   version 2.0.6

ellensp, 2020-08-09:
Please follow template and provide required Configuration files.

boelle, 2020-08-09:
@diver197 

just writing stm32f103re spi2 sd CARD, NO SD CARD version 2.0.6 does not say much (if anything)

also as @ellensp says read the template that was shown to you 

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

sjasonsmith, 2020-08-09:
I am going to go ahead and close this. There is no information to work with, and this is most likely a configuration issue.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-08-09 by sjasonsmith

diver197, 2020-08-09:
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5047651/configuration.zip)