MarlinFirmware/Marlin #18967: [BUG] Marlin Color UI: Z axis motion menu crashes Marlin

Miraculix200, 2020-08-09:
Using the Marlin Color UI on the Robin Nano v1.2 board, when selecting Movement -> Motion -> Z-Axis on the display, the screen goes blank and Marlin doesn't respond to serial input anymore (probably crashed). It happens both before and after homing.

Any idea what could cause this?

The compiler only shows these probably unrelated warnings:

```
.pio/build/mks_robin_nano35/src/src/gcode/gcode.cpp.o: In function `GcodeSuite::process_parsed_command(bool)':
/home/data/dev/3d/firmware/spro/bugfix-2.0.x/Marlin/Marlin/src/gcode/gcode.cpp:900: warning: undefined reference to `GcodeSuite::M993()'
/home/data/dev/3d/firmware/spro/bugfix-2.0.x/Marlin/Marlin/src/gcode/gcode.cpp:901: warning: undefined reference to `GcodeSuite::M994()'
```

My configuration: https://github.com/Miraculix200/Marlin-SapphirePro-2209-UART/tree/SapphirePro-2209-UART/Marlin

Miraculix200, 2020-08-10:
I found the problem. I've increased the buffers some time ago, to compensate SD card read errors. While this worked fine with the standard FSMC UI, it refused to compile/link because out of memory. Reducing the buffer sizes to half was not enough. Reducing them to half again fixed the problem.

boelle, 2020-08-10:
i think this is more a small feature request right?

rhapsodyv, 2020-08-10:
The color UI default buffer size is 19k. I changed it to 14k in nano boards. But it can be 8k with no problem. 

For SD IO errors, try putting this in your config:
```
#define SDIO_READ_RETRIES 16
```

closed 2020-08-12 by Miraculix200