MarlinFirmware/Marlin #18968: [FR] Touch_Mi_Retract (X Y and Z)

reefwarrior, 2020-08-09:
Hi,

Firstly, thank you for the work you have done on Marlin. It is still my 'go to' firmware.

I have a bed level probe; that deploys by going to Y_MAX + 6; however, to 'stow' it, it needs to go to certain coordinates, eg. X222 Y30 Z3 - but I have not been able to get this to work after either a G28 or G29. 

There is an option to run a script after G29, but that also does not appear to work.

I basically need the following to be done, in order to stow my probe:

G1 Y40 X222 Z20 F12000
G1 Z3 F5000
G1 Z20 F5000

(It does not use a magnet to attract the pin, it rather uses a plastic flag that deflects in an opening and 'locks' it in plate')

For reference the z-probe I have: https://www.thingiverse.com/thing:4182695

Any/all help appreciated.

Tinus

### Description

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

reefwarrior, 2020-08-09:
OK; I have gone and added code to two files: Configuration.h and probe.cpp

Lines of code added:
Configuration.h:

I don't know if somebody is perhaps willing to add this to the official code?

#define TOUCH_MI_PROBE
#if ENABLED(TOUCH_MI_PROBE)
  #define TOUCH_MI_MOVEMENT_RATE 10
  #define TOUCH_MI_STOW_Z_LIFT 30
  #define TOUCH_MI_STOW_X (X_MAX_BED + 2)
  #define TOUCH_MI_STOW_Y 30
  #define TOUCH_MI_STOW_Z 3
  #define TOUCH_MI_RETRACT_Z 0.5                  // Height at which the probe retracts
  #define TOUCH_MI_DEPLOY_XPOS (X_MAX_BED + 6)  // For a magnet on the right side of the bed
  //#define TOUCH_MI_MANUAL_DEPLOY                // For manual deploy (LCD menu)
#endif

probe.cpp
inline void run_stow_moves_script() {
    const xyz_pos_t oldpos = current_position;
    endstops.enable_z_probe(false);

    //do_blocking_move_to_z(TOUCH_MI_RETRACT_Z, MMM_TO_MMS(HOMING_FEEDRATE_Z));
    do_blocking_move_to_z(TOUCH_MI_STOW_Z_LIFT, MMM_TO_MMS(HOMING_FEEDRATE_Z*TOUCH_MI_MOVEMENT_RATE));
    do_blocking_move_to_x(TOUCH_MI_STOW_X, MMM_TO_MMS(HOMING_FEEDRATE_Z*TOUCH_MI_MOVEMENT_RATE));
    do_blocking_move_to_y(TOUCH_MI_STOW_Y, MMM_TO_MMS(HOMING_FEEDRATE_Z*TOUCH_MI_MOVEMENT_RATE));
    do_blocking_move_to_z(TOUCH_MI_STOW_Z, MMM_TO_MMS(HOMING_FEEDRATE_Z*TOUCH_MI_MOVEMENT_RATE));
    do_blocking_move_to(oldpos, MMM_TO_MMS(HOMING_FEEDRATE_Z*TOUCH_MI_MOVEMENT_RATE));
  }

Thank you for your time,

Tinus

closed 2020-08-09 by reefwarrior

AnHardt, 2020-08-09:
Your probe is much more like a Allen-key-probe than a Touch-Mi. Using the existent code for that should work.