MarlinFirmware/Marlin #18969: [BUG] Probe Z offset ignored after G28

HaernCalistinson, 2020-08-09:
### Bug Description

BTT GTR V1.0 board, BLTouch probe. After first configuration of the Z Offset (using M851 Zx.xx) the next G28 ignores the configured value.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5048274/Configuration.zip)

### Steps to Reproduce

Use the following process to configure the Z Probe Offset

1. M851 Z0 //remove existing offset
2. G1 Z0 //Move Z to 0
3. M211 S0 //Disable soft endstops
4. move bed to identify proper offset, in my case -5.2
5. M851 Z-5.2 //Set Z offset to new value
6. M211 S1 //Re-enable soft endstops
7. M500 //Save values
8. M501 //Load values
9. G28 //Home All
10. G1 Z0 //Move Z to 0

**Expected behavior:** [What you expect to happen]

At step 9, I should expect the Z value to display 5.2
At step 10 the bed should be just touching the nozzle, as set in step 4

**Actual behavior:** [What actually happens]

At step 7/8, after the values are saved and loaded to active, the Z location is updated to read Z0 as is expected. Moving Z to 10 and back to 0 returns Z to the proper location.

As soon as a home is performed, the Z offset is completely ignored and Z:0 is set back to the probe point (5.2mm below the nozzle).

#### Additional Information

I have been able to reproduce this in both 2.06 and bugfix-2.06.

I am able to work around this issue by setting the MANUAL_Z_HOME_POS setting to 5.2 and it works as expected, but the nozzle to probe setting should work as expected.

HaernCalistinson, 2020-08-11:
The most recent bugfix branch fixed this issue.

closed 2020-08-11 by HaernCalistinson

Plasmarobo, 2020-10-07:
I think there may have been a recent regression here.

Building with the latest bugfix (as of 10/6/2020) exhibits this issue: Z_HOMING properly sets the offset, but then it's erased immediately (set to whatever the homing offset is without accounting for probe offset).