MarlinFirmware/Marlin #18971: Build of 1.1.9 OK except it ignores End Stop Settings on Prusa i3 (Ooznest)

nigelbennee, 2020-08-09:


nigelbennee, 2020-08-09:
I did leave a comment but it has been swallowed. Someone please just give me a working link to a forum or similar where knowledge of Marlin exists. Thanks Nigel.

sjasonsmith, 2020-08-09:
Here you go! The Discord server is very active. I don't personally use any of the others. Most discussion will be around Marlin 2.0, it is becoming harder to find 1.1.9 help.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-08-09 by sjasonsmith