MarlinFirmware/Marlin #18973: [BUG] 2.0.5.3 Laser & servo (Bltouch) error in sanity check

chepo92, 2020-08-10:
### Bug Description

I'm configuring a Creality CR10S (mega 2560 board) with TTL laser, Bltouch and filament runout sensor, 
I can't figure out a wiring for everything that also compiles, I don't want to make any change to wiring every time i have to change to laser or 3D printing, so I want a dedicated pin for everything.
I have assigned in pin_RAMPS and RAMPS_creality:  
- Laser to pin 2 (PWM), 
- Bltouch servo to pin 11 (default Servo0 pin), and switch/sensor is pin 19 (Z max), 
- Moved runout to pin 15 (Y Max) (by default it is pin 2)

Also: 
- Enabled BL touch, set BLtouch pin to pin 19 
- Eabled Laser 
- Enabled filament runout detect
- Set num_servos to 1 

After changing the confings and start compiling I get a sanity check error: "Counter/Timer for SPINDLE_LASER_PWM_PIN is used by the servo system", so far i have read, enabling Bltouch enables servos and only leaves timer "A" pins unavailable, but pin 2 is on TIMER3B, 

Sanity check line is doing: 
#elif NUM_SERVOS > 0 && (WITHIN(SPINDLE_LASER_PWM_PIN, 2, 3) || SPINDLE_LASER_PWM_PIN == 5)
  #error "Counter/Timer for SPINDLE_LASER_PWM_PIN is used by the servo system."

I think maybe it's not doing the check correctly, as far as I understand the laser and servo [configuration](https://marlinfw.org/docs/configuration/laser_spindle.html) working, enabling certain number of servos should occupy a timer ¿"A"? and it's related pin and leave the rest of pins and timers for other uses, i.e. 
- with NUM_SERVOS = 0 only SERVO0_PIN is used (in ramps, it would be pin 11, attached to TIMER1A )
- with NUM_SERVOS = 1  SERVO0_PIN and SERVO1_PIN  are used (in ramps, it would add the use of pin 6, attached to TIMER4A)
- with NUM_SERVOS = 2  SERVO0_PIN, SERVO1_PIN and SERVO2_PIN  are used (in ramps, it would add the use of pin 5, attached to TIMER3A)
- with NUM_SERVOS = 3  SERVO0_PIN, SERVO1_PIN, SERVO2_PIN, SERVO3_PIN  are used (in ramps, it would add the use of pin 4, attached to TIMER0B)

Pin 2 and 3 are related to TIMER3B and TIMER3C, so it should not be a problem to even set any NUM_SERVOS > 0 , and use them for PWM, or is it ?, moreover I only need servo 0 for the BLtouch so I'm not even using Timer 3 for servos

Sorry for the length of the issue, any advice or comment would be apreciated

 

### My Configurations
I attach my configs and ramps/creality pins assignment 
[ramps.zip](https://github.com/MarlinFirmware/Marlin/files/5048623/ramps.zip)

### Steps to Reproduce

1. Explained in the description

**Expected behavior:** Compiles flawlessly

**Actual behavior:** sanity check error

#### Additional Information
I'm going to disable the sanity check line to see what happens, at least it allowed me to compile (will test if it works)

chepo92, 2020-08-10:
Current research:
- With BLtouch enabled, disabling the sanity check will pass compilation, but laser PWM in pin 2 won't work as it is tied to timer 3 and enabling BLtouch seems to attach all timers and servos
- Disabling the BLtouch will make the laser work
- With BLtouch disabled and define NUM_SERVOS 1, the laser won't work as it seems to also attach all possible timers 
- timers seem to be attached according to Marlin\src\HAL\AVR\ServoTimers.h, so it would be reasonable to comment line //#define _useTimer3 for testing purposes
- With BLtouch enabled and timer 3 disabled the laser seems to work

GMagician, 2020-08-13:
AFAIK Marlin usually doesn't use H/W pwm, so the relationship between timer and pwm handling timer is not 1:1.
Servo code uses timer 3 and 4 and pins 2/3 and 5 are on timer 3.

I think that a better test might be:
`#elif NUM_SERVOS > 0 && defined(_useTimer3) && (WITHIN(SPINDLE_LASER_PWM_PIN, 2, 3) || SPINDLE_LASER_PWM_PIN == 5)`

this new test will allow you to disable timer 3 and configure your machine

GMagician, 2020-08-13:
P.S your config is 020005, needs upgrate to 020006

boelle, 2020-08-13:
2.0.5.3?? 
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

GMagician, 2020-08-13:
@boelle this error will appears also in bugfix. Is how sanity is done and which H/W is used.

GMagician, 2020-08-13:
@chepo92 update your configuration and test PR

chepo92, 2020-08-13:
Tested Bugfix the same day, issue is still there 
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5070084/configs.zip)

@GMagician  I have made a change similar to your PR that works (but still needs improvements), I'll test yours but I think it may not work as  _useTimer3 will be defined when servos file is imported (I passed through that), so for now i commented out _useTimer3, see my branch: 
https://github.com/chepo92/Marlin/tree/CR10S5-BL-laser-fs

GMagician, 2020-08-13:
@chepo92 of course you have to disable _useTimer3 to let it works

GMagician, 2020-08-13:
@chepo92 I changed code, now AVR should release timers resources when not needed

chepo92, 2020-08-18:
It would be nice if setting NUMSERVOS to 0 it released all servo timers (including timer3 ), as BLtouch doesn't use timer3 by default
So you don't have to dig to servo code to release it
I had a similar solution and yours worked too (with de drawback of having to comment #define _useTimer3)

closed 2020-08-18 by chepo92

GMagician, 2020-08-18:
@chepo92 please try #19025, this will give priority to timer 4, then timer 3 is free and used only when "a lot " of servos are connected

PR #19011 (2020-08-14): [AVR] Let laser use some servo timers when they're not used
Relax sanity check when, in some custom configurastions, servo timer is freed

Fix #18973