MarlinFirmware/Marlin #18976: [FR] Write configuration to file(s) on SD

akamensky, 2020-08-10:
### Description

Currently Marlin will write configuration only to EEPROM (if board has one). Which leads to entire universe of headaches when trying to tinker with FW upgrades, tweaking settings here and there etc as often these will re-init EEPROM because of a single configuration being added or removed. This is especially frustrating as I will need to run bed leveling again and again as there is currently no way to back that up to anywhere.

I think the solution to those issues could be writing configuration as either plain G code or as .ini file to SD card, which would allow for backing configuration up, manually editing it (instead of sending a list of G code to alter them on printer) etc.

As a nice addition would be great to have bed leveling mesh stored in a separate file, which in turn could be backed up on it's own and restored when needed or manually altered (which is far easier than editing mesh via LCD menu). 

### Feature Workflow

Pre-requisites:

- SD support enabled in Configuration.h header file
- nothing else as SD library used in Marlin already supports writing files to SD

There are two options I can see for workflow:

1. A configuration switch that would disable EEPROM settings and store configuration to SD unconditionally
  i.e. something like
  ```
#ifdef SDSUPPORT
//#define SD_INSTEAD_OF_EEPROM
#endif
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Miraculix200, 2020-08-10:
I think you can enable SD card EEPROM emulation to save the config file to SD card, but I don't remember the config option. But the builtin EEPROM won't be used anymore then.

akamensky, 2020-08-10:
@Miraculix200 I've searched through and through in Marlin 2.0.5.4 and 2.0.6 and bugfix-2.0.x and have not found any such option. Searching on Google only finds above mentioned tickets that were closed without resolution.

FWIW, this ticket is to explore whether there is anything prohibitive or opposing this change. I could add this option on my free time.

Miraculix200, 2020-08-10:
#define SDCARD_EEPROM_EMULATION

akamensky, 2020-08-10:
Thanks. Found this in some board specific header files, but seems it is not in `Configuration[_adv].h` also seems no documentation on this option?

I will try it out to see whether that fulfills what I would like it to be.

Miraculix200, 2020-08-10:
It will probably work for all boards with SD card support. Some boards just don't have an EEPROM, so it's in t heir pins file

sjasonsmith, 2020-08-31:
@akamensky since you have not replied I am going to presume this worked out for you.
I am going to close this as a duplicate of another issue, which tracks the desire to back up settings to SD, even though SD is not the primary EEPROM mechanism.

sjasonsmith, 2020-08-31:
Duplicate of #12119

closed 2020-08-31 by sjasonsmith

Roxy-3D, 2020-08-31:
I like this idea.    Exporting the configuration to a file is pretty straight forward.    But we need to be careful on the import side of things.   It would be nice if we do more than reject the importation of the old configuration data if the firmware versions don't match.

I wonder if we could put some markers into the configuration data so we know that particular pieces of data are valid (and move them) but still reject pieces of data that have changed between the firmware revisions.   That probably is too hard to accomplish, but it might be worth some thought.