MarlinFirmware/Marlin #18977: [BUG] Odd behaviour with UBL

logiclrd, 2020-08-10:
### Bug Description

Forgive me in the event that this is a known issue, perhaps even already fixed, because my build of Marlin is a little bit behind. I do not have the liberty of updating it and trying a new version at this exact time, because I am observing the bug in an ongoing print that I don't want to interrupt and that I'm hoping will manage to produce acceptable results despite the bug. I have identified the exact upstream commit upon which my changes are currently based in the "Additional Information" section.

I have a printer with a 400x400 mm build plate, and I recently enabled UBL and measured a 15x15 mesh. I have included the mesh under "Additional Information" below.

I began a print that covers 350x350mm of the build plate -- almost the entire area. This print is proceeding very well in all areas except the very topmost area (largest Y values), where a strip is being treated very strangely, presumably by the UBL code. It's almost as if there is an abrupt end to the leveling mesh, and past that end, it does not interpolate linearly, but drops out suddenly, and every time the print crosses that boundary, Marlin raises the Z considerably, only to immediately lower it again. It's causing some artefacting that will almost definitely be visible in the finished product.

I have recorded a video of this behaviour: https://youtu.be/zdeDZaIIRdg

I also believe I have isolated the G code executing at the exact time of the video. I have included this in the "Additional Information" section as well.

This G code does not include any changes to the Z offset, so it seems likely all Z movement is the result of the UBL code. The coordinates in the G code range from (20, 20) to (380, 380), and I am printing with `M206 Y-9` to compensate for centering issues.

### My Configurations

[MarlinConfiguration.zip](https://github.com/MarlinFirmware/Marlin/files/5048985/MarlinConfiguration.zip)

### Steps to Reproduce

1. Enable UBL on a 400x400 bed with 20mm inset. Record a 15x15 mesh.

2. `M206 Y-9`

3. Begin a print that covers a square range from (20, 20) to (380, 380).

**Expected behavior:**

The print proceeds smoothly over the entire build area.

**Actual behavior:**

The print proceeds smoothly over the entire build area except for just a few millimeters at the upper extreme end of the Y range, where the bed starts wildly raising and lowering.

#### Additional Information

Marlin upstream base: 66676e6dd422f0518102be5375648a68d4de07c8

Video: https://youtu.be/zdeDZaIIRdg

UBL mesh:
```
>>> G29 T0
SENDING:G29 T0
Bed Topography Report:
    ( 20,380)                                                                                                      (380,380)
        0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
14 | -0.689  -0.784  -0.799  -0.699  -0.834  -0.670  -0.519  -0.567  -0.400  -0.394  -0.316  -0.070  -0.080  +0.105  +0.290
   |
13 | -0.504  -0.620  -0.631  -0.545  -0.643  -0.523  -0.410  -0.455  -0.266  -0.231  -0.171  +0.055  +0.056  +0.207  +0.359
   |
12 | -0.409  -0.566  -0.555  -0.451  -0.526  -0.436  -0.315  -0.346  -0.183  -0.158  -0.087  +0.145  +0.161  +0.334  +0.506
   |
11 | -0.249  -0.445  -0.418  -0.315  -0.439  -0.291  -0.200  -0.241 [-0.060] -0.084  +0.002  +0.214  +0.221  +0.393  +0.564
   |
10 | -0.205  -0.369  -0.357  -0.244  -0.329  -0.249  -0.149  -0.185  -0.040  -0.023  +0.016  +0.239  +0.219  +0.381  +0.544
   |
 9 | -0.085  -0.234  -0.263  -0.131  -0.225  -0.160  -0.050  -0.084  +0.072  +0.053  +0.101  +0.301  +0.290  +0.433  +0.575
   |
 8 | +0.049  -0.100  -0.110  +0.001  -0.116  -0.029  +0.055  +0.021  +0.173  +0.124  +0.187  +0.366  +0.355  +0.505  +0.655
   |
 7 | +0.079  -0.085  -0.044  +0.001  -0.119  -0.038  +0.053  +0.001  +0.156  +0.116  +0.157  +0.343  +0.317  +0.449  +0.580
   |
 6 | +0.127  -0.038  -0.069  +0.039  -0.090  -0.008  +0.061  +0.009  +0.156  +0.109  +0.143  +0.310  +0.268  +0.397  +0.527
   |
 5 | +0.154  -0.006  -0.029  +0.064  -0.087  +0.005  +0.058  -0.011  +0.121  +0.066  +0.086  +0.275  +0.215  +0.337  +0.460
   |
 4 | +0.145  -0.018  -0.046  +0.016  -0.131  -0.050   0.000  -0.077  +0.048  +0.010  +0.015  +0.199  +0.120  +0.229  +0.337
   |
 3 | +0.207  +0.048  +0.002  +0.055  -0.084  -0.023  -0.003  -0.070  +0.065  +0.007  +0.016  +0.167  +0.100  +0.202  +0.305
   |
 2 | +0.166  -0.030  -0.046  +0.012  -0.135  -0.079  -0.054  -0.161  -0.030  -0.059  -0.119  +0.046  -0.072  +0.021  +0.115
   |
 1 | +0.060  -0.156  -0.195  -0.118  -0.283  -0.206  -0.166  -0.291  -0.170  -0.259  -0.249  -0.118  -0.205  -0.126  -0.048
   |
 0 | -0.046  -0.255  -0.295  -0.264  -0.429  -0.385  -0.376  -0.475  -0.374  -0.425  -0.471  -0.344  -0.462  -0.404  -0.345
        0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
    ( 20, 20)                                                                                                      (380, 20)
```

UBL bug video G-code: [UBL bug video.gcode](https://github.com/MarlinFirmware/Marlin/files/5049014/UBL.bug.video.gcode.txt)

thisiskeithb, 2020-08-10:
> Marlin upstream base: 66676e6

Marlin is updated daily, so it's best to try the latest [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) so we're not chasing potential bugs that may have been fixed since April.

logiclrd, 2020-08-10:
I have just diffed 66676e6 to [bugfix-2.0.x](/MarlinFirmware/Marlin/tree/bugfix-2.0.x) and found no substantive differences in code that could be related to motion or bed levelling. There were plenty of differences, but they were all in one of these categories:

* Non-code.
* HAL, LCD, LEDs, DAC, pins, external UI and other hardware.
* G-code command implementations.
* Not related to motion or bed leveling in any way.
* Non-semantic changes (adding `const`, switching between `#ifdef` and `TERN` macros,

In particular, `ubl_motion.cpp` is tagged as having a change, but the change is a single line, making the license URL in the comment at the top use `https://` instead of `http://`.

I also looked in particular detail at the `planner.cpp` and `planner_bezier.cpp` files.

I am not able to just merge latest and rebuild right now, because this code is actually running the build where this odd movement has been observed. But, I am reasonably convinced that there haven't been code changes to the parts responsible for this behaviour.

thisiskeithb, 2020-08-10:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

logiclrd, 2020-08-10:
Well, at the current rate I won't be able to do that for about a week. This print will be running until then. I'm disappointed that there's zero possibility of any investigation while the print is running that might reveal a solution that could prevent further damage to the print itself. But, once this print is complete, I'll reapply my configuration changes onto the latest `bugfix-2.0.x` and make a test pattern to reproduce the issue in that area.

logiclrd, 2020-08-30:
I have upgraded to the latest `bugfix-2.0.x` (as of 2020-08-15), and the problem continues. I have made a new video showing the behaviour with a test pattern I created to try to verify my UBL mesh.

Video: https://youtu.be/ziZTLfE1tbE

New mesh (parameters are slightly different, but the overall shape looks pretty much the same):
```
    ( 25,375)                                                                                                      (375,375)
        0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
14 | -0.298  -0.474  -0.480  -0.408  -0.581  -0.509  -0.406  -0.543  -0.439  -0.396  -0.404  -0.235  -0.221  -0.165  +0.093
   |
13 | -0.050  -0.269  -0.335  -0.219  -0.403  -0.333  -0.198  -0.360  -0.224  -0.198  -0.246  -0.046  -0.041  -0.015  +0.233
   |
12 | -0.024  -0.189  -0.269  -0.149  -0.328  -0.256  -0.149  -0.262  -0.201  -0.115  -0.215  +0.043  +0.017  +0.085  +0.328
   |
11 | +0.115  -0.102  -0.160  -0.062  -0.238  -0.206  -0.100  -0.201  -0.100  -0.064  -0.130  +0.029  +0.048  +0.157  +0.370
   |
10 | +0.195  +0.019  -0.088  +0.020  -0.186  -0.128  -0.050  -0.176  -0.078  -0.032  -0.060  +0.101  +0.105  +0.172  +0.388
   |
 9 | +0.233  +0.057  -0.022  +0.048  -0.121  -0.104  -0.015  -0.114  -0.018  +0.016  -0.021  +0.145  +0.151  +0.224  +0.438
   |
 8 | +0.382  +0.186  +0.074  +0.165  -0.020  +0.014  +0.114  -0.022  +0.070  +0.103  +0.072  +0.199  +0.206  +0.262  +0.471
   |
 7 | +0.424  +0.210  +0.113  +0.195  +0.011  +0.028  +0.125  +0.023  +0.098  +0.120  +0.065  +0.207  +0.220  +0.278  +0.466
   |
 6 | +0.422  +0.202  +0.123  +0.198  +0.013  +0.030  +0.120  -0.005  +0.081  +0.112  +0.045  +0.189  +0.158  +0.223  +0.418
   |
 5 | +0.473  +0.273  +0.180  +0.258  +0.052  +0.082  +0.151  +0.008  +0.095  +0.107  +0.040  +0.159  +0.149  +0.184  +0.332
   |
 4 | +0.443  +0.235  +0.145  +0.232  +0.037  +0.035  +0.130  -0.020  +0.046  +0.063  -0.017  +0.073  +0.076  +0.133  +0.332
   |
 3 | +0.460  +0.259  +0.166  +0.235  +0.028  +0.051  +0.129  -0.030  +0.042  +0.035  -0.098  +0.079  +0.040  +0.065  +0.264
   |
 2 | +0.468  +0.287  +0.173  +0.249  +0.038  +0.068  +0.139  -0.012  +0.016  -0.019  -0.056  +0.071  +0.034  +0.021  +0.175
   |
 1 | +0.398  +0.135  +0.049  +0.118  -0.070  -0.094  -0.035  -0.209  -0.093  -0.094  -0.206  -0.066  -0.069  -0.078  +0.111
   |
 0 | +0.266  +0.045  -0.076  -0.025  -0.224  -0.196  -0.121  -0.295  -0.259  -0.204  -0.330  -0.215  -0.282  -0.283  -0.108
        0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
    ( 25, 25)                                                                                                      (375, 25)
```

qwewer0, 2020-08-31:
> Video: https://youtu.be/ziZTLfE1tbE

I too experienced this at the back and right side of my bed, just like on the video. But mine might does that weird z movement in the other direction, so it squishes the line more to the bed.
@logiclrd Is it raises your bed at the back?

SKR Mini E3 v1.2
Ender 3
Latest Bugfix-2.0.x

logiclrd, 2020-08-31:
In my case, it seems to temporarily go to higher Z value, which for my printer means physically the bed moves down. It moves a lot, too, way more than one layer's worth. The original video linked in the issue description shows this behaviour from a different angle.

qwewer0, 2020-08-31:
Ok, now I see. I have `INVERT_Z_DIR false`, so that could be why in my case the nozzle moves closer to the bed, while it moves away for you. But not sure if this discovery is any use for us...

tanaes, 2020-09-14:
Just chiming in, that I'm experiencing the same behavior using a recent bugfix as base (d9ad8caf0281398dfe960b4babb4e46512e18b18; my firmware [here](https://github.com/tanaes/Marlin/tree/Genius-bugfix)). 

Unlike @logiclrd's mesh, which has negative offsets at the border of the behavior, mine has positive offsets:

```
Recv: Bed Topography Report:
Recv: 
Recv:     (  1,219)                                      (219,219)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.083  +0.083  +0.090  +0.044  +0.130  +0.185  +0.245
Recv:    |
Recv:  5 | +0.080  +0.080  +0.090  +0.025  +0.089  +0.111  +0.197
Recv:    |
Recv:  4 | +0.076  +0.076  +0.131  +0.007  +0.048  +0.036  +0.148
Recv:    |
Recv:  3 | +0.065  +0.065  +0.067  -0.005  +0.039  +0.061  +0.154
Recv:    |
Recv:  2 | +0.093  +0.083  +0.073  +0.012  +0.063  +0.065  +0.189
Recv:    |
Recv:  1 | +0.110  +0.080  +0.049  +0.001  +0.047  +0.048  +0.121
Recv:    |
Recv:  0 | +0.141 [+0.123] +0.105  +0.025  +0.043  +0.005  +0.033
Recv:         0       1       2       3       4       5       6
Recv:     (  1,  1)                                      (219,  1)
```

Whenever the nozzle travels near the rightmost margin of the print bed, it abruptly drops to zero, as if no offsets are being applied there.

qwewer0, 2020-10-15:
I still experiences sudden jumps in Z height at the back and right side of the bed with UBL.

Latest Bugfix-2.0.x
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5382734/Configuration.zip)

tanaes, 2020-10-26:
@qwewer0 curious, where is your origin defined? Mine is slightly outside my 220x220 printable area, at -5,0. I'm wondering if the nozzle jumping is occurring when the nozzle is going beyond x_max + x_min, or 215.

qwewer0, 2020-10-26:
@tanaes Not sure, but I only tested to the end of the mesh, and not to the end of the bed.
But I don't think marlin lets the nozzle to go beyond the travel limits (X/Y/Z_MIN_POS, X/Y/Z_MAX_POS)

logiclrd, 2020-10-26:
In my case, my bed is 400x400 and `MESH_INSET` is 25, so my grid runs from (25, 25) to (375, 375). I'm seeing the odd behaviour with points that are inside the bed area but outside of the mesh.

logiclrd, 2020-10-26:
But, I should add, not _all_ points that are inside the bed area and outside of the mesh.

ManuelMcLure, 2020-10-26:
That's because unlike Bilinear, UBL does not extrapolate outside the mesh. The proper way to use UBL is to make the mesh the size of the bed and set PROBING_MARGIN to avoid probing outside the (25,25)/(375,375) area. Then use `G29 P3` to fill in the points that were not probed with reasonable values.

qwewer0, 2020-10-26:
I got this problem within the UBL mesh, so...

logiclrd, 2020-10-26:
Hmm, so if I don't want to probe the outer 25mm on all sides, I can't get better than a 13x13 mesh? Why is there a 15-point hard upper limit on the number of points per axis?

ManuelMcLure, 2020-10-26:
The limit is because a mesh is one of the biggest users of EEPROM. You can definitely get a 15x15 mesh - it's just that you can't _probe_ all of it. `G29 P3` will fill the unprobed points with reasonable values by extrapolation, and then you can adjust them with test prints.
This is better discussed on the Marlin Discord instead of in a bug, however. https://discord.gg/n5NJ59y

qwewer0, 2020-10-27:
> it's just that you can't probe all of it.

I have 15x15 UBL mesh on "235x235" bed with PROBING_MARGIN/MESH_INSET of 10 and all of the mesh points are probed automatically by the probe, because my X/Y_MIN_POS, X/Y_MAX_POS is allows it.

logiclrd, 2020-11-06:
I decreased my `MESH_INSET` to a value outside of which I probably wouldn't feel comfortable asking the printer to print anyway, and instead relied on `X_MAX_POS` and `Y_MAX_POS` to determine which points are actually probe-able, using `G29 P3` to extrapolate values for the ones it couldn't reach, and am not observing the behaviour I documented originally, with the mesh now encompassing the X/Y range of the print.

sjasonsmith, 2020-11-28:
@ManuelMcLure and @logiclrd is there actually an issue here, or can this be closed?

It seems like this is likely due to a misunderstanding regarding the UBL mesh, and everything works fine when the print is confined to the mesh boundaries?

qwewer0, 2020-11-28:
I can't seem to reproduce the issue on the latest bugfix-2.0.x.
Within or outside of the UBL mesh, all the Z axis corrections are smooth without any sudden jumps. Maybe there were some PRs that changed something with UBL.

Side question: Is there a way to see what extrapolated values does UBL use outside of the mesh?

ManuelMcLure, 2020-11-28:
As far as I know UBL does not extrapolate outside the mesh.

qwewer0, 2020-11-28:
@ManuelMcLure Thanks, didn't know that.
Then it uses the closes mesh value?

ManuelMcLure, 2020-11-28:
This is probably better discussed on Discord since there doesn’t seem to be a bug here.

rhapsodyv, 2020-12-10:
So, this can be closed?

boelle, 2020-12-10:
i would say yes.... no response from logiclrd in over a month
everyone else seems not able to reproduce and also mentioned better discussed on discord

logiclrd, 2020-12-10:
I can't immediately test it, because my printer is presently broken 😢 but I did do a couple of print runs after reworking the settings to not use MESH_INSET, and I'm pretty sure I did not observe the strange behaviour.

qwewer0, 2020-12-10:
Could not reproduce it on the latest bugfix, but I would wait for @logiclrd's printer to see if the problem still coccus.

sjasonsmith, 2020-12-24:
It seems very likely that this has been addressed by #20538. Please re-test with the bugfix-2.0.x branch and close this issue if it is no longer a problem.

tanaes, 2020-12-24:
Unfortunately away from my printer for about a month, but I will definitely check as soon as I can get back to it. This sounds like exactly the thing!

logiclrd, 2021-01-28:
Just had another encounter with this behaviour, and can confirm that is almost certainly "by design", but I think the design could perhaps be improved. Basically, if it reaches a point where the mesh isn't defined (either past its boundaries, or a point that hasn't been sampled), it treats it as though it's a zero, causing a very abrupt change in Z. This is almost certainly undesirable. Perhaps there should be a better way to handle this? Improved feedback, improved default behaviour? In the print I have running right now, I had just re-measured the grid, because I added hardware to my hot end and re-flashed the firmware, and the new hardware restricts the range slightly. Because of this restriction, the entire line of points at max X were unsampled, because it can't get the probe there. I saw it behaving really oddly in the print every time it got to the right edge, and actually issued `G29 P3` _while it was printing_ and it resolved the issue. Since I expect there will never be a scenario where someone _actually wants_ the Z to fall off a cliff in certain regions, maybe it would make sense to support something like an implicit `G29 P3` w.r.t. to how it actually applies the mesh while running?? Or, well, the `G29 P3` calculations didn't seem to take enough time to run for the running print to even notice them, but if they are more expensive calculations, perhaps a simpler approximation could be used, along with a warning to the operator on the screen and/or serial to let them know they're working with a defective bed? Or a check and a warning when e.g. `M24` runs to let you know that you're starting a print with an incomplete mesh?

closed 2021-03-14 by github-actions[bot]