MarlinFirmware/Marlin #18979: [BUG] (short description)

gopgop9000, 2020-08-10:
### Bug Description

When using the Ender 3 I tend to get a Thermal Runaway error after starting a print in which the printer was preheated. 
Asked on a discord and two other users have the same problem (both run a 32bit board, the skr1.3, whereas I use the Ender 3 V2's 32bit board)
I run all stock hardware (besides the Ender 3 V2's 32bit board, which mine came with for some reason)
I run Marlin 2.0.6 (Ender 3 Pro V1.5 config preset)
Both of the other run some version of Marlin 2.0, one does does not know specific version at this time of writing, other's is 2.0.1.
One of the other users also stated that manually setting the temperature instead of using the Preheat PLA function will not reproduce the problem.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5049329/Configuration.zip)

### Steps to Reproduce

1. Select Preheat PLA 
2. Let it fully preheat
3. Start printing something

**Expected behavior:** Print as it should 

**Actual behavior:** Thermal runaway happens sometime after it starts (happens within the first few layers for me)

chepo92, 2020-08-10:
I sometimes have the same issue with default configs, you have to increase the thermal runaway timings and adjust the pid as the layer fan cools the hotend and it is not strong enough to re-heat in time before TR triggers

gopgop9000, 2020-08-10:
Alrighty, thank you for the information.

closed 2020-08-10 by gopgop9000