MarlinFirmware/Marlin #18980: [BUG] G35, Different Z Travel Movement Speeds

qwewer0, 2020-08-10:
### Bug Description

Sending `G35` to the printer, for the 1st measured point (right after the XYZ homed position) the ([BLTOUCH_HS_MODE](https://github.com/MarlinFirmware/Marlin/blob/fd9c6983d30f70171d9d9a9c55962689fa4cdd85/Marlin/src/gcode/bedlevel/G35.cpp#L124)) z travel movement speed is 4mm/s, just like my `HOMING_FEEDRATE_Z` and `Z_HYBRID_THRESHOLD`, so it runs on StealthChop, but for the other three points the z travel movement speed is 5mm/s, and runs on SpreadCycle (so I was able to hear the difference). Tested it with `Z_HYBRID_THRESHOLD 5`, and with that setting all the z travel movements are running on StealthChop mode. Only the "to point" z travel movement speed is the problem, the actual probings are done with the right speeds.
The problem isn't the fact that it runs on the louder SpreadCycle mode, but for 3/4 z travel moves are done with 5mm/s speed, which is (theoretically) different from any of my settings.

### My Configurations

SKR Mini E3 v1.2, Ender 3
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5049889/Configuration.zip)

### Steps to Reproduce

0. Settings: 

- BLTOUCH_HS_MODE ([z movements](https://github.com/MarlinFirmware/Marlin/blob/fd9c6983d30f70171d9d9a9c55962689fa4cdd85/Marlin/src/gcode/bedlevel/G35.cpp#L124) for measured points)
- HYBRID_THRESHOLD enabled
- Z_HYBRID_THRESHOLD 4 (for easier speed/stepping mode determination)
- Z_SAFE_HOMING
- HOMING_FEEDRATE_Z  (4*60)
- ASSISTED_TRAMMING enabled

1. Send G35 to the printer
2. Listen to the stepping modes for the z movements

**Expected behavior:**
`G35` should use the same (`HOMING_FEEDRATE_Z`) z travel movement speeds for all points.

**Actual behavior:**
The z travel movement speed for the first measured point (after `G35` homes) is different from the z travel movement speeds for the other three measured point.

sjasonsmith, 2020-08-17:
> The z travel movement speed for the first measured point (after G35 homes) is different from the z travel movement speeds for the other three measured point.

I don't use this feature, but I looked at the code to see what might cause this.

The moves you are concerned about are while moving between points, so the move is in X, Y, and Z?
Moves between points are done at `XY_PROBE_SPEED`. If there is a Z component to that move, it will move whatever speed necessary to make the entire move at `XY_PROBE_SPEED`.

Most likely you have a larger Z move when moving between points than when moving to the first point. This requires a higher speed on the Z axis, crossing your hybrid threshold.

qwewer0, 2020-08-17:
If there isn't anything weird in the code, then it might be like that.

Thanks for the enlightenment.

closed 2020-08-17 by qwewer0