MarlinFirmware/Marlin #18983: [FR] Forward extruder-stepper action on other extruder-stepper

Novotny-Martin, 2020-08-10:
### Description

I have 3 extruders  E0,E1,E2  which feed thru 3in1 adapter  E3. E3 is directly on hotend. I need forward all movement from E0-E2 to E3. It will work like MMU2, but without separate MCU and complex mechanics.
### Feature Workflow

#### Additional Information

GMagician, 2020-08-10:
**OT** @Novotny-Martin any detailed info about how your extruder looks like?

Novotny-Martin, 2020-08-10:
> 
> 
> **OT** @Novotny-Martin any detailed info about how your extruder looks like?

> 
> 
> **OT** @Novotny-Martin any detailed info about how your extruder looks like?

I plan to use Bulldog extruder 
https://d2t1xqejof9utc.cloudfront.net/screenshots/pics/5c5c23950c07b2a514eeed256876d766/original.jpg

AnHardt, 2020-08-12:
Most of us already know how a single extruder is locking like.

> I have 3 extruders E0,E1,E2 which feed thru 3in1 adapter E3. E3 is directly on hotend.

is what we have problems with to imagine.

GMagician, 2020-08-12:
@AnHardt I suspect he will use E0, E1 and E2 to feed E3 but once E3 is feeded if other steppers don't release filament then an almost sure issue will occur, it's almost impossible to use 2 steppers and feed the same amount of filament.

sjasonsmith, 2020-08-12:
I work with one printer that uses two motors on each filament. It has plenty of problems but actual filament movement hasn’t seemed to be one of them.

sjasonsmith, 2020-08-12:
Actually now that I think about it more, maybe I do have feed issues on that printer. I hadn’t put a lot of thought into the competing motors being the cause of any issues, but maybe it is.