MarlinFirmware/Marlin #18988: [BUG] Neopixel indexing when 2 strings used.

SparkyDan555, 2020-08-10:
### Bug Description

When using the "I" variable of M150 alongside NEOPIXEL2_PIN the indexes are duplicated across both strings. This makes it impossible to index a specific pixel. May I suggest that the first index of the second strip is after the last index of the first strip. This would allow the user to reference all the indexes of the first strip followed by all the indexes of the second strip, allowing such individual control.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/5053587/Archive.zip)

### Steps to Reproduce

1. Enable NEOPIXEL_LED and NEOPIXEL2_PIN
2. Compile and upload
3. Send "M150 I5 R255"

**Expected behavior:** [What you expect to happen]
The 6th LED of the first string lights up red.

**Actual behavior:** [What actually happens]
The 6th LED of BOTH strings lights up.

AnHardt, 2020-08-10:
> When using the "I" variable of M150 alongside NEOPIXEL2_PIN the indexes are duplicated across both strings. 

That is exactly what it's ought to do - filling the double amount of pixels without doubling the transfer time. If you just want more different pixels connect them in series.

SparkyDan555, 2020-08-10:
This is an option but greatly complicates wiring as a data return wire would be needed. In the SKR series boards it is easier to use some auxiliary pins such as servo or power detect for a second neopixel strip. Timing should be unaffected as 2 smaller strips split between 2 pins is still the same number of pixels as one strip in series on just one pin. It's the same number of pixels to be parsed and stored in RAM etc.

SparkyDan555, 2020-08-10:
The current functionality effectively makes neopixel 2 pin redundant as the data signal from one pin could simply be split in parallel between 2 strips. This dual pin function is only helpful if the second neopixel pin can output a independent data stream.

ellensp, 2020-08-11:
This is a FR not a bug. It works as intended.

ellensp, 2020-08-11:
Have a play with the above PR. Set #define NEOPIXEL2_INSERIES true in Configuration.h

SparkyDan555, 2020-08-13:
I finally got round to testing this PR tonight and it seems to work well. I think this is the simplest way to implement this as the length of the second strip doesn't need to be known, the indexes are simply added onto the end of the first one. Good work and I hope this can be merged.

ellensp, 2020-08-14:
Has just been merged, is now part of marlin bugfix.

closed 2020-08-14 by ellensp