MarlinFirmware/Marlin #18990: Marlin 2.0.6 fails on Anycubic Trigorilla Pro

rq3, 2020-08-10:
I have successfully installed Marlin 2.0.5.3 on an Anycubic Trigorilla Pro board. This required:
1) adding "board_build.core=maple" to the trigorilla_pro environment in platformio.ini
2) modifying the z height parameter from 427mm to 456mm in configuration.h to get probing to function
3) physically rotating the LCD touchscreen 180 degrees

Under Marlin 2.0.6 the compile is successful and loads on to the Trigorilla Pro. However, the display background has artifacts, the touch pad locations have moved to the top of the screen, and swapped left to right, and probing no longer functions.

I have attempted to compile using both my earlier configuration.h and configuration_adv.h, and also the ones under 2.0.6 "examples:delta:anycubic:predator", and have attempted many modifications to flip the display and enable probing.

The files involved are obviously based on a release by SHX, who used an SK board, and loaded firmware other than Marlin.

I'd be happy to help in any way I can. I have modified my printer so that no jumper swapping on the Trigorilla is required to flash firmware, only the flip of a switch.

ellensp, 2020-08-11:
You need to try this on bugfix 2.0.x. These issues may have already been addressed.

rq3, 2020-08-11:
I did try this on the latest bugfix. Won't compile, and throws errors regarding the touch button calibration and positions.

boelle, 2020-08-11:
> I'd be happy to help in any way I can.

submitting a PR is the best way to go, and the fastest

rhapsodyv, 2020-08-11:
> button

It throw erros if the calibration values isn’t defined. If you define those values, do you get any error?

rq3, 2020-08-11:
> 
> 
> > I'd be happy to help in any way I can.
> 
> submitting a PR is the best way to go, and the fastest

Thanks, boelle. It seems to be down to aligning the touch areas with the screen display. If I could rotate the screen image 1180 degrees, but NOT the touch areas, all would be well. I don't understand the units used for calibration and offsets of the touch areas in configuration.h, so I'm struggling in a random fashion.

rq3, 2020-08-11:
> 
> 
> > button
> 
> It throw erros if the calibration values isn’t defined. If you define those values, do you get any error?

I have no idea how to do that! But I'm willing to try! I'm assuming this will require editing of the xtp2406.cpp file?

rhapsodyv, 2020-08-11:
> I have no idea how to do that! But I'm willing to try! I'm assuming this will require editing of the xtp2406.cpp file?

You define the calibration values in your Configuration file. Try one existing. Some boards have defaults values, like chitu and nano boards.

ellensp, 2020-08-12:
Please provide config files and error logs..  "Won't compile, and throws errors" isn't very useful.

ellensp, 2020-08-12:
If the errors your getting are these... 
```
Marlin/src/lcd/touch/xpt2046.cpp:115:22: error: 'XPT2046_X_CALIBRATION' was not declared in this scope
Marlin/src/lcd/touch/xpt2046.cpp:116:22: error: 'XPT2046_X_OFFSET' was not declared in this scope
Marlin/src/lcd/touch/xpt2046.cpp:117:22: error: 'XPT2046_Y_CALIBRATION' was not declared in this scope
Marlin/src/lcd/touch/xpt2046.cpp:118:22: error: 'XPT2046_Y_OFFSET' was not declared in this scope
```
You simply need to un-comment the following from Configuration.h

```
  //#define XPT2046_X_CALIBRATION 12316
  //#define XPT2046_Y_CALIBRATION -8981
  //#define XPT2046_X_OFFSET        -43
  //#define XPT2046_Y_OFFSET        257
```

rq3, 2020-08-12:
> If the errors your getting are these...
>
> ```
> Marlin/src/lcd/touch/xpt2046.cpp:115:22: error: 'XPT2046_X_CALIBRATION' was not declared in this scope
> Marlin/src/lcd/touch/xpt2046.cpp:116:22: error: [...]

ellensp, many thanks for that! I'm no code wizard, but your comment allowed me to get a successful compile. After that, I managed to flip the display (rotate 180), in the configuration_adv.h file, and then realign the touch area calibration by changing the Y offset to 75. I am printing, have some fine tuning and calibration to do, but all is well!

I really appreciate the immediate and succint help available here. You have all reinforced my faith in humanity.

closed 2020-08-12 by rq3