MarlinFirmware/Marlin #18995: ["EVENT_GCODE_SD_STOP is now EVENT_GCODE_SD_ABORT. Please update your Configuration.h."] (in bugfix-2.0.x)

mclemmon, 2020-08-11:
### Bug Description

EVENT_GCODE_SD_STOP is now EVENT_GCODE_SD_ABORT. Please update your Configuration.h

### My Configurations

Configurations were default from the Configurations-bugfix-2.0.x\config\examples\Creality\Ender-3\BigTreeTech SKR Mini E3 2.0 repo

### Steps to Reproduce

1. replace configuration.h and configuration_adv.h to marlin folder
2. try to build

**Expected behavior:** successful build

**Actual behavior:** does not build, giving error of "EVENT_GCODE_SD_STOP is now EVENT_GCODE_SD_ABORT. Please update your Configuration.h"

#### Additional Information

EVENT_GCODE_SD_STOP is not in the configuration.h file, but instead in the configuration.adv.h file. Line 1142 in configuration.adv.h should be changed from #define EVENT_GCODE_SD_STOP "G28XY" to #define EVENT_GCODE_SD_ABORT "G28XY"

ellensp, 2020-08-11:
This was added to the code 10 hours ago...  config have not been updated yet. 
Just do what it says.

closed 2020-08-11 by ellensp

mclemmon, 2020-08-11:
I did, I just thought I'd bring it to others attention..

doug-scott, 2020-09-20:
> I did, I just thought I'd bring it to others attention..

Thanks for doing that.  Yesterday I had no issues building the firmware, today I got that error after making some changes to my configuration.h file.  Still running the same source, did not touch configuration_adv.h.  Changing STOP for ABORT fixed it.  How can this happen?