MarlinFirmware/Marlin #18996: [BUG] Marlin mode working only when printer is powered via USB (and not by main power)

digant73, 2020-08-11:
### Bug Description
Marlin mode is correctly working only when the printer is powered via USB cable (e.g. connecting a laptop to the USB port in the main board).
When the printer is powered on via main power, Marlin menus disappears and I see only a black background.
Furthermore, if the main power is switched off and the printer remains powered via USB only, Marlin menu remains not working. I need to reset (e.g. via the reset button on the TFT) the printer in order the Marlin menu is displayed again using USB powering.
I already tested the flat cable I used and it is OK (continuity is OK on all the wires). I also used other cables with same results.
I also shielded the flat cable using copper tape all around the flat cable and connecting the shield to DC 0V on the power supply.
I also used different delay in Configuration.h such as:

#define ST7920_DELAY_1 DELAY_NS(0)
#define ST7920_DELAY_2 DELAY_NS(400)
#define ST7920_DELAY_3 DELAY_NS(0)

#define ST7920_DELAY_1 DELAY_NS(0)
#define ST7920_DELAY_2 DELAY_NS(250)
#define ST7920_DELAY_3 DELAY_NS(250)

and many more
But, nothing fixed the problem

### Steps to Reproduce
1. [First Step]
power on the printer via USB. Marlin menus are correctly displayed
2. [Second Step]
power on the printer via main power. Marlin menus disappears (a black background is displayed)
3. [and so on...]

**Expected behavior:** [What you expect to happen]
Marlin menu works also when the printer is powered on via main power

**Actual behavior:** [What actually happens]
Marlin menus are not displayed if the printer is powered on via main power

#### Additional Information
main board: MKS GEN L
TFT: MKSTFT28

boelle, 2020-08-11:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

ellensp, 2020-08-11:
This is clearly an electrical issue. nothing to do with firmware.

closed 2020-08-11 by ellensp