MarlinFirmware/Marlin #18999: [FR] Support for the Tenlog D3Hero display (and other Nextion displays?)

cbaugher, 2020-08-12:
### Description
This is intended to be a discussion about adding support for the Tenlog Nextion display used in the D3Hero printers and probably some others.  This could maybe be expanded to include all Nextion displays.

Some work has already been started on this, so feel free to join in here if you have anything to add.  (you know who you are ;-)

Also if this is not the place for this, let me know and we'll take it elsewhere.

ellensp, 2020-08-12:
Duplication of https://github.com/MarlinFirmware/Marlin/issues/7754

closed 2020-08-12 by ellensp

studiodyne, 2020-08-14:
Idea: Make a donation group , and ask @thinkyhead  to make the module + a github for the marlin OEM firmware
Not infinite time wasted / not expensive if we are a big group, 

Marlin really need , to have a DISPLAY , not 1000 differents displays with differents screen not complete ! , impossible to use a  without the old lcd, because  there is always something that missing , and that need to learn LINUX/PYTHON/JAVA and more to just add a screen with ' linear advance '!

Nextion is the solution