MarlinFirmware/Marlin #19005: [FR] Sensitivity autotune

hackebike, 2020-08-12:
Some drivers like TMC 2209 or TMC 2226 have StallGuard for sensorless homing or probing.

It would be great to have a autotune feature to find the sensitivity value to set for the X/Y/Z driver's sensitivity.
I have found that same driver like 2209 have different values for X/Y/Z so it's always difficult to find the correct value.
Right now cannot find a good value for 2226 on Y/Z but for X found to be 65 !

An autotune would help a lot instead of going through all 256 values for each axis.

Der-WU, 2020-08-15:
Nice FR.
But if it possible?
I Hope....

hackebike, 2020-08-15:
Not sure but it would be great! I am trying to find a value for sensitivity on TMC 2226, found one for X but cannot find one for Y and Z. On 2209 found all. But it's a pain doing it manually!

ellensp, 2020-08-15:
not really possible... how can it tell that it is smashing into the end of travel when sensitivity is wrong...or it just hitting a slightly rough patch and it not really at the end of travel It can't. 
   
It also depends on a number of factors...  driver current, rate of movement, type of stepper driver... for some combinations there is no value that will work.

hackebike, 2020-08-15:
[StallGuard](https://www.trinamic.com/technology/motor-control-technology/stallguard-and-coolstep/) works to detect those situations, so it should be possible to do an autotune detection.

ellensp, 2020-08-15:
Only when its already configured. catch 22

hackebike, 2020-08-15:
Where is the code StallGuard is defined ?

tesla128, 2021-02-01:
Turn off StallGuard
Move backwards 10mm (all numbers configurable)
Turn on StallGuard
Move forward 10mm.  decrease sensitivity if stalls early.
Repeat until 10mm movement is consistent
Move forward until stopped
decrease sensitivity 10%
Move back and forward again
If stopped again, stop
If axis is not at end-stop, re-run tuning with minimum sensitivity from previous run or EEPROM value if reset

Might not be perfect, but seems like it would give a good starting point over poke-and-hope