MarlinFirmware/Marlin #19006: [BUG] Tronxy xy-2 Pro | Printing suddenly Stops. Extruder heating stays on!

hahn23, 2020-08-12:
[I flashed the new Marlin release to my Tronxy XY-2 Pro with Mainboard CXY-V6191017

I started to print an dinosaur model. But there were some problems with.

Shortly after the first start the Printer stops mid print. The Display is unresponsive to every touch. Everything is just "freezed".
Bad thing: The heating is not stopped! After I turned the Printer off and on again it showed nearly 300 C°  and the nozzlde burned a hole into my heatingbed... 

I thought it could be a Problem with the G-Code, so I generated it this time with Prusaslice Software. (Settings for Marlin firmware).

After I copied that to my SD-Card I startet a new try.

After 2:30 H (The model was nearly completed....) It happened again. It stopped mid print and the heating was raising again.

If I wouldn't have stayed near the Printer all the time... This could realy end in an bad situation...

Any thoughts, what I could try? I added an ZIP file containing the Configs I used and the G-Code of the Dino.

[marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5064585/marlin.zip)

boelle, 2020-08-13:
> new Marlin release

what version?

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

hahn23, 2020-08-14:
the firmware I used was compiled from the bugfix-2.0.x branch

rhapsodyv, 2020-08-14:
Are you using Linear Advance with TMC?

hahn23, 2020-08-14:
I checked my adv config.... I have //#define LIN_ADVANCE so it is turned off. Should I use it?!

rhapsodyv, 2020-08-14:
> I checked my adv config.... I have //#define LIN_ADVANCE so it is turned off. Should I use it?!

No, it's just to confirm.

sjasonsmith, 2020-08-14:
This is _probably_ a duplicate of #18358.

There is a workaround somewhere in that thread you can try, but you will have some reading to find it in that other thread.

If you determine it does sound like the same issue this one should be closed as a duplicate.

hahn23, 2020-08-14:
> This is _probably_ a duplicate of #18358.
> 
> There is a workaround somewhere in that thread you can try, but you will have some reading to find it in that other thread.
> 
> If you determine it does sound like the same issue this one should be closed as a duplicate.

Jason your hint was realy a good one! I tried #Patch1 of your linked workaround and I was able to sucessfully print my Model without any issues. Before I tried the very last release of the Firmware and compiled everything new... But it got stuck after about 1 H.

Then I saw your link and I deleted the complete build folder on my HDD and did everything completely again. 
And so far.... it realy looks like this was the solution so far. 

Thank you all very much for helping me. Great job

hahn23, 2020-08-14:
My happiness was a bit to early.... Tried an other Model... it startet printing it... I went away.... It freezed and continued heating. 

Unfortunaly that was it. The hotend is burned and my room full of smoke. The Display only shows "Not reeched E1" Printer stopped Please restart.

3 Day old printer -.-

boelle, 2020-08-14:
> 3 Day old printer -.-

China Quality at its best, prob a FET shorted closed

sjasonsmith, 2020-08-14:
> China Quality at its best, prob a FET shorted closed

This comment doesn't provide any value. I linked to another issue which documents well firmware issues regarding STM32F1 boards that can cause these problem. It has nothing to do with board quality or where it was designed/manufactured.

I am going to close this issue as being a duplicate of #18358. I'm sorry there isn't a better solution for this right now @hahn23, but this does appear to be the same problem.

sjasonsmith, 2020-08-14:
Duplicate of #18358

sjasonsmith, 2020-08-14:
Actually reading back @boelle might be right. The issue I linked probably would not have shown a temperature error on the screen.
It seems this started as a duplicate of the other issue, at this point it may have evolved into a hardware failure. Either way, I don't think there is a software bug to track aside from #18358.

closed 2020-08-14 by sjasonsmith