MarlinFirmware/Marlin #19008: [BUG] MKS Robin E3D inductive sensor Auto Bed levelling strange movements

ekef, 2020-08-13:
### Bug Description

I have changed my Ramps 1.4 with MKS Robin E3d recently. I used to use ramps 1.4 and same configuration (probe, switches , marlin) perfectly. Then i have installed new motherboard and uploaded marlin, i can't do auto bed levelling.(Probe sensor works for Z axis.) Auto bed levelling sometimes does not start at all. The table goes up and stops before the sensor sees it then stops. Sometimes the first probe point works but stops when it passes to the second. Sometimes the first probe works, it switches to the second and does nothing to the third. I tried everything in settings. I just dealt with this one day. I cannot use Marlin 2.0.5 because the card is not supported. I can only use 2.0.6. I have tried last Marlin bugfix.
I added some videos.

### My Configurations

My printer is hypercube core xy. 
My inductive sensor is LJ12A3-4-Z/BY (I have soldered LM7805 for drop voltage to 5v.)
My marlin config 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5066482/Marlin.zip)

**Expected behavior:** do auto bed levelling

**Actual behavior:** Does not do auto bed levelling

#### Additional Information

https://drive.google.com/file/d/1yzOrPfa4xGuKEJH0Vx2J48LP0XC2bvLp/view?usp=sharing4
https://drive.google.com/file/d/1yuwnKUKt47He9YJFKqAQoFxE6qjmWEWf/view?usp=sharing

boelle, 2020-08-13:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

ekef, 2020-08-15:
It works, thanks.

closed 2020-08-15 by ekef