MarlinFirmware/Marlin #19014: Case mismatch in "PID_Debug_Flag" variable causes Marlin compilation to fail

Bougakov, 2020-08-13:
There is a variable called `PID_Debug_Flag` defined in `temperature.cpp` - it is set if `PID_DEBUG` is enabled.

Unfortunately, in `M303.cpp` it is written in lowercase, as `pid_debug_flag`. This case mismatch causes compilation to fail.

Solution - replace 4 occurrences of `pid_debug_flag` in `M303.cpp` with `PID_Debug_Flag`.

Bug was originally posted by @jelred in https://github.com/MarlinFirmware/Marlin/issues/17446#issuecomment-633667364 but closed without resolution.

GMagician, 2020-08-13:
Try bugfix. In current code variable is `pid_debug_flag` so maybe it's already fixed

boelle, 2020-08-13:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

ellensp, 2020-08-14:
In current bugfix this is all in lowercase ie "pid_debug_flag" throughout the code.

closed 2020-08-14 by ellensp