MarlinFirmware/Marlin #19015: Ender 3 with the new v4.2.7 board

fizzi-bubble, 2020-08-13:
hello guys,
I'm using the latest marlin 2.0.6 and i'm trying to get the new v4.2.7 board working with my ender 3, because this is now the board you get when you are ordering a silentboard for your printer. The ender 3 v2 has an other display than the v1 so i was trying  to get this working for two days now.
only defining "cr10_stockdisplay" in the configuration.h doesn't work.

boelle, 2020-08-13:
could you add [FR] to the start of the title?

btw... use bugfix instead as its updated faster
[`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)

badouuuuuu, 2020-08-13:
Hello, i already created an issue in a wrong github : https://github.com/MarlinFirmware/Configurations/issues/185

@boelle We are not able to upgrate to 2.0.6 with the 4.2.7 board on Ender 3 v1
Axes are not working, and we got "Homing Failed".
I tried several times with differentes config files, release 2.0.6 & bugfix

@fizzi-bubble what is your trouble ? You don't see anything on your screen ? Can describe your issues ?

fizzi-bubble, 2020-08-13:
@badouuuuuu 
with this firmware you just linked i could get the temps and the steppermotors running. my biggest problem is now that i have a bltouch installed but it has no function. only the selftest at the start of the printer is running. that's why i need a running config so i can enable the bltouch.

badouuuuuu, 2020-08-13:
> @badouuuuuu
> with this firmware you just linked i could get the temps and the steppermotors running. my biggest problem is now that i have a bltouch installed but it has no function. only the [...]

Try this one : https://mega.nz/file/CHpmTKTQ#azdbn0DjvbTmZYQmmjxD_L0lp3Ya4iZQ9Onbi1mWmgY
It's a 2.0.6 i built for you with BL Touch activate, just to check, but i think you'll have homing failed

ellensp, 2020-08-14:
Firstly you need MOTHERBOARD BOARD_CREALITY_V4
Then you need to enable CR10_STOCKDISPLAY
then under the #define CR10_STOCKDISPLAY line you need to add either #define RET6_12864_LCD or #define VET6_12864_LCD depending MPU on your board is a RET6 or a VET6

ellensp, 2020-08-14:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-08-14 by ellensp

badouuuuuu, 2020-08-14:
It's not about the screen, but the steppers, we can't moves axies, and homing is failed with 2.0.6

I'm not an expert but something need to be fix in the pins files ? I'll post a bug report
4.2.7 are weird, why only a firmware named "TMC2225" works on this board

ellensp, 2020-08-14:
@badouuuuuu attempt to hijack anothers issue denied.  This issue from fizzi-bubble was exclusively about the screen.
I am working on moving your issue to correct place and reopening it.

badouuuuuu, 2020-08-14:
@ellensp are you on meth ?

I closed the old ones, there is no "attempt/hijack/denied issue", i just wanted to inform about the new 4.2.7 board trouble.
Learn to be kind with people, it's not our jobs.

Thanks.